christianhaitian/arkos #878: There is no selection to uae4arm core in Retroarch

MAN-biker, 2023-12-25:
Game console: R36S (RK3326).
Distro Version: ArkOS 2.0 (12222023)
Retroarch: 1.16.0 (14cb3732eb) 03 Oct. 2023
GCC (8.3.0) 64-bit

In ArkOS menu there are 3 options for AMIGA emulation:

1. STANDALONE
   - AUTO
2. **Retroarch**
   - PUAE2021
   - PUAE
3. Retroarch**32**
   - **UAE4ARM**
  
I NEED exactly **UAE4ARM** (NOT PUAE2021 or PUAE, because of it's BEST of all with good performance for most games).
But it;s not possible to select **Retroarch\UAE4ARM** (Retroarch**32**\UAE4ARM only), but in this case, there is NO CD-32 and CDTV support at all in Retroarch**32**:
AUTO
A500
A600
A1200

But if I run **Retroarch** directly and load **UAE4ARM** core, I see ALL of options, including **CDTV** & **CD32**:
AUTO
A500
A600
A1200
**CDTV
CD32**

So, if **Retroarch** HAVE **UAE4ARM** core (and it's BETTER, than UAE4ARM core in Retroarch**32**, why i CAN'T select it in ArkOS menu?!

christianhaitian, 2023-12-25:
Because there was never a request to add support for the 64bit version of uae4arm to the Emulationstation menu.

MAN-biker, 2023-12-25:
Why?!
It seems like the BEST Amiga\CD32 emulation core for such low\middle end devices (by performance).
And why Retroarch32\UAE4ARM core missing CDTV & CD32 emulation?

In any case, is it possible to add these features?

Thanks in advance.

christianhaitian, 2023-12-26:
No need to get so excited.  Everyone doesn't play Amiga.  It's not as popular as other consoles and some are probably satisfied with uae4arm on the 32bit side because it satisfies what they want the emulator to do.  Some prefer puae2021.  Either way support in emulationstation for those emulators have been added as they've been requested where possible.  I will include that in the next update.  The standalone amiberry emulator does not have virtual keyboard support because it hasn't been updated since last year or so.  I'll review the possibility of updating that emulator as virtual keyboard support was added back in January of this year.

lauren7ino, 2024-01-16:
Hi Christian!
Do you have any idea when this modification to Amiga support will be available?
Thanks, and keep up the excellent work! :)

christianhaitian, 2024-01-16:
Sometime this month

christianhaitian, 2024-01-27:
This has been addressed with today's update.

closed 2024-01-27 by christianhaitian