christianhaitian/arkos #879: Scummvm stark_surface_fill.vertex Shader

eth3real, 2023-12-26:
Hello, I'm using an RG353V (V2 display) unit with Arkos, fully updated, and trying to launch The Longest Journey in ScummVM standalone, version 2.8.0git. It's failing with the following error:

`ERROR: Could not open shader stark_surface_fill.vertex!!`

Other ScummVM titles are running with no issue. I see that the `shaders` folder is present at `/opt/scummvm/extra/shaders`, but this one is missing. Any suggestions? Thanks in advance!

eth3real, 2023-12-26:
I went and grabbed that file and `stark_surface_fill.fragment` from the main [ScummVM Git repo](https://github.com/scummvm/scummvm/tree/master/engines/stark/shaders), dropped them into `/opt/scummvm/extras/shaders/` and everything is running fine now. Weird that they were missing, though, right?

christianhaitian, 2023-12-26:
Interesting.  I'll make sure to include that in the next update.  Thanks for reporting this.

christianhaitian, 2024-01-27:
This has been addressed with today's update.

closed 2024-01-27 by christianhaitian