christianhaitian/arkos #880: Request: emmc support for installing ArkOS on rg353m devices to internal memory rather than to only sdcard on TF1

dexx0008, 2023-12-27:
Would it be possible to add support for internal emmc memory for arkos? Currently it does not show emmc memory when you ssh into the device.

Using other means I was able to flash arkos img to internal memory; however, at first boot (with no sdcards installed) while It does boot to the ArkOS logo it doesn't proceed to install. After a short bit of time it switches from arkos logo to a blank screen and moves no further.

christianhaitian, 2023-12-27:
It is something I'd be interested in as I have no use for Android on the 353 devices and would prefer to make use of the internal emmc for my own use case at least.  How did you flash ArkOS to the internal emmc?

closed 2023-12-29 by christianhaitian

christianhaitian, 2024-01-05:
Thanks for this info.  In reviewing this and conversing with the JELOS team, I was able to get a solution for supporting this and it seems to be working well.  See here for more information. https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG353M#q-can-i-load-arkos-onto-the-internal-emmc-storage

dexx0008, 2024-01-23:
Sorry for the late reply, thank you for the update. - this worked
perfectly. Successfully installed ArkOs on internal. Just need to move some
games to the new sd card in slot 2 and do a little setup, i'll leave sdcard
1 blank

Have a spot where i can donate something for your efforts as a thank you?

christianhaitian, 2024-01-23:
You're welcome.  The main page of the wiki has a donation link at the top and bottom.  No donation is needed but if you'd like to donate, I appreciate it.  Enjoy!