christianhaitian/arkos #881: ambernic rg353v not booting

Dennis11111110, 2023-12-28:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki? yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ? yes

### Describe the Issue (If applicable)
A clear and concise description of what the bug is. so i flashed the sd card with the image and i ejected and once i put it in the rg353v it gets stuck on the Arkos loading and dosent get passed that.

### How can the issue be reproduced? (If applicable)

### What device are you using?
rg353v

### What version of ArkOS are you on? i am using the one from here that says Gdrive 
- You can find this during the initial boot process right before the Emulationstation loading screen

### Anything else you'd like to include that may help to review this issue or feature request? i have reflashed and rebooted a milion times but nothing is working i have used multiple sdcards and sd card reader and i have had it charged for all attempts plzz help.

Dennis11111110, 2023-12-28:
also sorry to add am not sure if i have v1 or v2 but when i try and change the names i am missing one of the files that i am supposed to change

christianhaitian, 2023-12-28:
See here for more info about this issue you're experiencing.  Not an ideal solution but since I can't replicate the issue on my end, it's the best I can offer until I receive a unit with this issue. https://github.com/christianhaitian/arkos/issues/837

closed 2023-12-28 by christianhaitian