christianhaitian/arkos #883: Games still display on RG353V screen even though external HDMI cable is attached

EnsignRutherford, 2023-12-29:
I attached an external HDMI cable to my RG353V.  EmulationStation displays fine on the external monitor however when games run they still run on the RG353V's screen.  Is this expected behavior?

christianhaitian, 2023-12-29:
Does not sound even feasible as once the hdmi cable is plugged into the unit and is rebooted in order to display through a connect monitor, the internal display is disabled.  If anything, nothing should display if it can't display via the hdmi connection.

EnsignRutherford, 2023-12-29:
Ok this time I rebooted and it works.  But kinda.  I tried to play Shinobi and the game plays, but the scenes where it shows the map before play starts are black.  I'll test more to see if maybe it's the cable (brand new) or something else.

christianhaitian, 2023-12-29:
Sounds like a cable issue to me.  Make sire you’re using a good cable and not one with an adapter.  See here for more info.  https://github.com/christianhaitian/arkos/issues/828

closed 2023-12-29 by christianhaitian