christianhaitian/arkos #886: Ps1 Emulation

gallardworth, 2023-12-31:
Hi, I would like to report an issue. In my device, R36s, Ps1 emulation is very laggy on nfs 3 hot pursuit. Whatever settings I try it is laggy like its 3-5 fps. I use your last updated version 12222023. Someone in YouTube states that on 09292023 update there is no lag in spite of their previous update version. Could you please make a suggestion or a fix to solve this problem? Thanks.

christianhaitian, 2023-12-31:
Did you update the core in retroarch32?

gallardworth, 2023-12-31:
I have just tried but the result is the same. Nfs3 hp is so laggy that no
one can play.

şunu yazdı:

christianhaitian, 2024-01-01:
Then try a different emulator such as duckstation or swanstation and see if the results are better with one of those emulators.

closed 2024-01-01 by christianhaitian

gallardworth, 2024-01-01:
Duckstation or swanstation do not even start the game. Screen being black
then returning back to the game selection page.

şunu yazdı:

christianhaitian, 2024-01-01:
Sounds like you’re missing a bios for the ps1 or you have a bad rom