christianhaitian/arkos #887: ArkOS not installing in Anbernic RG353V

djinn011, 2023-12-31:
Hi!

I'm trying to install the ArkOS on my newly purchased Anbernic RG353V with the 10152023 version. I followed properly the installation guide from formatting my 128GB Ultra SanDisk SD card to writing it using Win32DiskImager. However, it seems to be not installing properly. Is there any solution to this?

christianhaitian, 2023-12-31:
what exactly is happening?

djinn011, 2023-12-31:
Most of the time it gets stuck at the ArkOS logo. Sometimes it gets to the next page with the build number and four loading dots then gets stuck. Sometimes it gets to the partition part. I've had successfully installed it 2 times but then hangs during navigation of the UI. When I restart it, it gets stuck again in the ArkOS logo and never get to load the OS again so I tried re-installing which still won't work.

It's pretty random.

christianhaitian, 2023-12-31:
You have a unit that has issues with ArkOS.  See here for a workaround: https://github.com/christianhaitian/arkos/issues/831#issuecomment-1794064668.

closed 2023-12-31 by christianhaitian