christianhaitian/arkos #888: ArkOS can't run zip file but jelos can.

Boybangs, 2024-01-01:
I can extract it before putting to SDcard that's not a problem but neo geo games can't do that. 

I tried to put the same the set of roms with the same bios to jelos  and it's run fine.

My device : Powkiddy rgb30

christianhaitian, 2024-01-01:
Maybe you didn’t copy the games correctly. Zip files for neo geo work fine on my end along with other systems as long as zip files are supported for the systemper the documentation.

closed 2024-01-01 by christianhaitian

Boybangs, 2024-01-01:
Could you tell me how to copy correctly?

On Jelos and ArkOS I used the same copy method but ArkOS can't run zip file.

christianhaitian, 2024-01-01:
For neo geo, i place the zip files along with the bios file in neogeo folder then run them per the ArkOS Emulators and Ports info section of the wiki here: <https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#neo-geo>

Boybangs, 2024-01-01:
I did that and it can't run.

christianhaitian, 2024-01-01:
Try updating your cores via the retroarch menu.  Other than that, I don’t know what your issue is as I can run neo geo games fine without issue.

Boybangs, 2024-01-01:
Not only neo geo games but other console like nes or SNES if use as zip extension it can't run.

Thanks for helping me.

christianhaitian, 2024-01-01:
Are you using a Mac?

Boybangs, 2024-01-01:
I'm using a pc, windows 10 .

christianhaitian, 2024-01-01:
odd.  I have no issues using .zip files with any the systems you mention and many others don't either with ArkOS.  Something is wrong with your install but I can't offer any more ideas on what the issue could be.  Sorry.

Boybangs, 2024-01-01:
Ok thanks for helping me.

Boybangs, 2024-01-02:
One more question, what brand of sdcard are you using?

christianhaitian, 2024-01-02:
Samsung and Sandisk sd cards are all I use for gaming on these devices.

Boybangs, 2024-01-02:
Ok thank for your reply.

Boybangs, 2024-01-04:
I come to inform you that's the problem has solved because I changed SDcard to SanDisk. It's can run zip file now. 😁