christianhaitian/arkos #889: Rgb10s stick Invested

ElSoplanucas, 2024-01-01:
### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
Rgb10s stick remains reversed after selecting the option Set controls to RGB10s mode
### How can the issue be reproduced? (If applicable)

### What device are you using?
Rgb10s 

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
Ark OS 2.0 (12222023)

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2024-01-01:
In the boot partition, is there a `rk3326-odroidgo2-linux-v11.dtb.rgb10s` file in there?  You should be able to access that partition on the sd card from either a Windows, Mac or Linux machine

ElSoplanucas, 2024-01-01:
> In the boot partition, is there a `rk3326-odroidgo2-linux-v11.dtb.rgb10s` file in there? You should be able to access that partition on the sd card from either a Windows, Mac or Linux machine

The file is there but when i select Set controls to RGB10s mode in Bart says the trigger have been activated for oga 1.1
[image]

christianhaitian, 2024-01-01:
That's odd.  Delete the 'rk3326-odroidgo2-linux-v11.dtb' and manually copy the `rk3326-odroidgo2-linux-v11.dtb.rgb10s` to `rk3326-odroidgo2-linux-v11.dtb`.  Make sure you copy the `rk3326-odroidgo2-linux-v11.dtb.rgb10s` to `rk3326-odroidgo2-linux-v11.dtb`, not rename it.  Then boot the device.

ElSoplanucas, 2024-01-01:
> That's odd. Delete the 'rk3326-odroidgo2-linux-v11.dtb' and manually copy the `rk3326-odroidgo2-linux-v11.dtb.rgb10s` to `rk3326-odroidgo2-linux-v11.dtb`. Make sure you copy the `rk3326-odroidgo2-linux-v11.dtb.rgb10s` to `rk3326-odroidgo2-linux-v11.dtb`, not rename it. Then boot the device.

It works, thanks for your help

closed 2024-01-02 by ElSoplanucas