christianhaitian/arkos #891: Enterprise 128 Support Missing

EnsignRutherford, 2024-01-02:
Hi,

Happy New Year!

According to the wiki the Enterprise 64 and 128 are supported however there's no entry in es_systems.cfg and the core is not in the retroarch's core folder.

Is this a documentation error or was it intended to be available?

ER

christianhaitian, 2024-01-02:
Happy New Year to you as well.  Did you update to at least the 12/15/2023 update of ArkOS?

EnsignRutherford, 2024-01-04:
Not yet.  I will when I get home so I can make a backup of the SD CARD.

EnsignRutherford, 2024-01-05:
I bit the bullet and did the upgrade.  Looks like it worked well.

christianhaitian, 2024-01-05:
Yea those should be done from time to time to get new OS updates.  Those updates don't include retroarch or retroarch32 core updates.  You an do those directly from within retroarch.  You can see the changelog for OS updates for your particular device here: https://github.com/christianhaitian/arkos/wiki/Changelogs

closed 2024-01-05 by christianhaitian

EnsignRutherford, 2024-01-06:
Gotcha.   I'm thinking about adding TRS-80 Coco / XRoar siupport.  This device is amazing and your OS takes it over the top!