christianhaitian/arkos #894: ArkOs doesnt Boot in RGB30

kenavru, 2024-01-05:
As in topic, just got new black RGB30 from aliexpress, JelOs works fine. ArkOS doesnt even boot. Got black screen and green led - like with no SD card at all.
Tried 6 different cards, even one that came with console, RGB30 doesnt even try to boot it. 
XZ uncompressed.
Tried on windows with win32diskimager and that one sugested by you.
Also DD with linux - same effect. Partitions looks fine
[image]
Even swaped dts file with no OC and ones from jelos - no change.

It looks like Boot partition is not found by RGB30.
the one from jelos has some additional flag ?
[image]

christianhaitian, 2024-01-05:
Sounds like you may have a unit that just won't boot ArkOS without reducing the clock speed, however, those usually at least show the ArkOS boot logo first.  From your description, it seems like the unit acts like it doesn't even see the sd card at all.  Are you using a Samsung or Sandisk sd card?  You mention 6 different sd cards but never mention the make of any of them.  

As another test, you can try extracting the uInitrd from attached file and copy it to the BOOT partition overwriting the existing one and see if that makes a difference.  It forces the unit to boot from the root partition.

[uInitrd_force_root.zip](https://github.com/christianhaitian/arkos/files/13843185/uInitrd_force_root.zip)

kenavru, 2024-01-05:
sandisk ultra 32GB
kioxia exceria 16GB U1 A1
Kioxia from RGB30 jelos
Kingston canvas 128GB
Samsung EVO 32GB
all the same

prepearing fresh ark card and will try what you sugested.

I may also try to copy data from 4/5 partition over jealos partitions ...

kenavru, 2024-01-05:
nope force_root does nothing.

christianhaitian, 2024-01-05:
The fact that you don't even get the splash screen makes me think something else is going on here.  Have you tried USB IT to flash the sd card?

https://www.alexpage.de/usb-image-tool/download/

kenavru, 2024-01-05:
yep, same thing. Also redownloaded and DD it od linux. Same.

kenavru, 2024-01-05:
hah, booted, got arkOS LOGO.

only when i copied content of 3rd partition over jealos as on screen.

so problem is with partition table of ArkOs image

christianhaitian, 2024-01-05:
Interesting.

kenavru, 2024-01-05:
ok, not solved - still doesnt boot after restart ... 
just blind-luck boot was that time probably xD
cant make it boot again

kenavru, 2024-01-05:
or it may be this ...
https://www.reddit.com/r/PowKiddy/comments/18117q1/rgb30_arkos_black_screen_on_reboot/
https://youtu.be/KjoHK29iZFk?t=327

kenavru, 2024-01-05:
totaly no idea ... will pass for now, I'm reflashing for half a day ;D

christianhaitian, 2024-01-05:
Understood.  Thanks for taking time to try and figure it out.  Enjoy

closed 2024-01-05 by christianhaitian

kenavru, 2024-01-05:
my last 2 suspects are:
1. some lose bits of random data on not clean SD card - that prevents rk3566 to boot.
2. it has totaly different uboot than Jelos

christianhaitian, 2024-01-06:
ArkOS does use the JELOS uboot as default.

christianhaitian, 2024-01-06:
For Deep Sleep, you can switch to the Anbernic uboot if the unit can reliably support it.

kenavru, 2024-01-06:
> ArkOS does use the JELOS uboot as default.

well they look 80% different for me, as just extracted em ;) will check out tomorrow ;p

christianhaitian, 2024-01-06:
you may be talking about the gpt file system structure.  The uboot used between JELOS and initial boot of ArkOS are the similar:

JELOS: https://github.com/JustEnoughLinuxOS/rk356x-uboot
ArkOS: https://github.com/christianhaitian/rk356x-uboot

kenavru, 2024-01-06:
ok, swaping uboot/layout from jelos to ark gives nothing, so can remove point 2.

christianhaitian, 2024-01-06:
See if replacing the `rk3566-OC.dtb` file in the BOOT partition with the one within the attached zip file allows it to boot and is stable.  It reduces the max speed of the soc to 1.4Ghz.  Not ideal for emulation above ps1.
[rk3566-OC.zip](https://github.com/christianhaitian/arkos/files/13850044/rk3566-OC.zip)

kenavru, 2024-01-06:
nope, i checked many other dtb - no change. It doesnt get to that point to load it.

christianhaitian, 2024-01-06:
I can't tell from this pic. https://raw.githubusercontent.com/christianhaitian/arkos/main/pics/rgb30/20230913_153334_resized.jpg

kenavru, 2024-01-06:
looks like it doesnt ;( will be hard to debug this problem

ILikeIneine, 2024-01-10:
I got the same problem. I can only see arkos logo when I flashed the os image to the sd card at the first time I launched the rgb30. And then, after it set the partition and auto reboot, the screen will always be black with green led lights on. Nothing would happen:(

kenavru, 2024-01-13:
@ILikeIneine managed to get it work ?

ILikeIneine, 2024-01-21:
> @ILikeIneine managed to get it work ?

No, I installed jelos. Have you solved it?

ILikeIneine, 2024-01-22:
Yeah! I fixed it in a rather weird way...

I tried another tf card and flashed the arkos, it successfully booted. I back up the whole tf disk to a image file. (I do this with disk genius) and this completely dump the structure of the workable   
arkos.

Then I restore the image to the previous tf card that wouldn't properly work. It just mirrored all the disk structures and files. But when I insert it to rgb30, logo appeared and successfully booted.

All I've done was just dump the tf card that could *install arkos and launched correctly* to the tf card that would *fail to install and boot arkos* . If u need, I can share my self-made image.

[image]

[info: back up disk to image](https://www.diskgenius.com/manual/backup-disk-to-image-file.php)
[info: restore disk from image file](https://www.diskgenius.com/manual/restore-disk-from-image-file.php)

kenavru, 2024-01-23:
@ILikeIneine you can put image somewhere.

vivre30, 2024-03-23:
Hi @ILikeIneine @kenavru 
Did your RGB30 is the new one with the colored ABXY?

I got problem as well, but mine still can boot to ArkOS by pressing reload/reset button
If not, will be blank after ArkOS logo.
here is my vid on reddit : https://www.reddit.com/r/PowKiddy/comments/1bilrsb/rgb30_arkos_wont_installboot_without_press/

and can you share your working image @ILikeIneine to me, thanks

kenavru, 2025-02-27:
today i took my rgb30 from box to test it out ;p On kioxia Sd card current version od arkos didnt boot as usual.
Next took red Aiwa 64GB SD, and i runed, installed and worked fine, until turn off. Then it didnt turn on as usual :D black screen.
No hope in my rgb i see.

christianhaitian, 2025-02-27:
> today i took my rgb30 from box to test it out ;p On kioxia Sd card current version of arkos didnt boot as usual. Next took red Aiwa 64GB SD, and i runed, installed and worked fine, until turn off. Then it didnt turn on as usual :D black screen. Still no hope with my rgb i guess.

Did you use the latest image available?

kenavru, 2025-02-27:
yes.

Also checked out os that wasnt present back then, Rocknix, works fine, also rumble works out of box at boot ;)
turned off/unpluged SD etc still works just fine ;)
So i will probably just gift it out with this OS.
ArkOS is the only one that doesnt like my rgb30.