christianhaitian/arkos #895: Feature Request - Add TRS-80 Coco to ArkOS emulators

EnsignRutherford, 2024-01-07:
Hi,

I've been trying to add support following the examples of how other emulators are currently setup, but to no avail.

I've tried all mame and mess cores to try and load a game disk in a zip file.  I have the roms required in the bios folder, but retroarch starts and stops with no error message.

Can you assist?

ER

christianhaitian, 2024-01-07:
How is it supposed to work?

EnsignRutherford, 2024-01-07:
Not sure... I'm researching how it works on other platforms.

EnsignRutherford, 2024-01-08:
So I've made some progress... 
Placed two xml files from MESS into the hash folder of mame, and put the bios files in the proper folder.
This command is used to execute:
/usr/local/bin/retroarch -v -L /home/ark/.config/retroarch/cores/mess_libretro.so /roms2/coco/nebula.zip

This loads up everything ok and identifies the proper emulation "coco" but fails to find the rom file.
[INFO] RetroArch 1.16.0 (Git 14cb3732eb)
[INFO] Saving files in content directory is set. This overrides other save file directory settings.
[INFO] Saving save states in content directory is set. This overrides other save state file directory settings.
[INFO] === Build =======================================
[INFO] Capabilities: ASIMD
[INFO] Version: 1.16.0
[INFO] Git: 14cb3732eb
[INFO] Built: Oct  3 2023
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Core]: Loading dynamic libretro core from: "/home/ark/.config/retroarch/cores/mess_libretro.so"
[INFO] [Environ]: SET_VARIABLES.
[INFO] Saving files in content directory is set. This overrides other save file directory settings.
[INFO] Saving save states in content directory is set. This overrides other save state file directory settings.
[INFO] [Overrides]: Redirecting save file to "/roms2/coco/nebula.srm".
[INFO] [Overrides]: Redirecting save state to "/roms2/coco/nebula.state".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: SYSTEM_DIRECTORY: "/roms2/bios".
[libretro INFO] SYSTEM_DIRECTORY: /roms2/bios
[INFO] [Environ]: CORE_ASSETS_DIRECTORY: "/home/ark/.config/retroarch/downloads".
[libretro INFO] CONTENT_DIRECTORY: /home/ark/.config/retroarch/downloads
[INFO] [Environ]: GET_SAVE_DIRECTORY.
[libretro INFO] SAVE_DIRECTORY: /roms2/coco/
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.
[INFO] [Content]: Content loading skipped. Implementation will load it on its own.
[ERROR] [Environ]: GET_VARIABLE: mame_nobuffer - Invalid value.
[libretro INFO] Joystick map: disabled
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[libretro INFO] Starting game from command line:/roms2/coco/nebula.zip
[libretro INFO] ARGUV[0]=/roms2/coco/nebula.zip
[libretro WARN] Driver nebula not found -1
[libretro WARN] Game not found: nebula
[libretro INFO] Game name: coco, Game description: Color Computer
[libretro INFO] Game name: coco, Game description: Color Computer
[libretro INFO] System found: coco
Adding monitor screen0 (640 x 480) a:1.333333
[libretro INFO] SOURCE FILE: ../../../../../src/mame/trs/coco12.cpp
[libretro INFO] PARENT: 0
[libretro INFO] NAME: coco
[libretro INFO] DESCRIPTION: Color Computer
[libretro INFO] YEAR: 1980
[libretro INFO] MANUFACTURER: Tandy Radio Shack
[libretro ERROR] bas10.rom NOT FOUND (tried in coco)
[libretro ERROR] Fatal error: Required files are missing, the machine cannot be run.
[libretro INFO] RES:2

bas10.rom is in /roms2/coco, /roms2/bios, and roms2/bios/coco but still doesn't find it.

Still investigating.  I'm close.

EnsignRutherford, 2024-01-08:
Got it... moved bas10.rom to coco.zip in /roms2/coco and it works!

christianhaitian, 2024-01-08:
Sweet.  I'll check this out when possible.

EnsignRutherford, 2024-01-08:
[image]

EnsignRutherford, 2024-01-08:
Okay much like Dungeons of Daggorath, abandon hope all ye that enter the dungeon... or something like that.  I've tried several cartridges to test and the high resolution screens are mangled.  It's not playable unless I'm doing something incredibly wrong.  No errors are being generated.

EnsignRutherford, 2024-01-08:
Ok, got it working after doing something silly.  Changed the name of the rom folder from 'coco' to 'coco3' and all the cartridges I've tested work and I was even able to play Canyon Climber to complete the first level!  I'll update this post with the configuration details shortly.

christianhaitian, 2024-01-08:
Fantastic!

christianhaitian, 2024-01-08:
Man the nostalgia is strong here.  The Tandy CoCo3 was my first computer.  First computer game I got for it was Rampage.  I'll be testing this soon.  I'm hoping if it's stable enough to include this with the next update I release either later this week or next.

EnsignRutherford, 2024-01-08:
I wrote a bunch of utilities sold by Spectrum Projects back in the day... 40k Basic, Print Spooler, Multipak-Crack, etc.  In 2000 I wrote Return of Coco, a coco 1/2 emulator so I could play Dungeons of Daggorath in Windows.  Nostalgia indeed!!

christianhaitian, 2024-01-08:
You're using mess so unless mess has the keyboard feature built-in, it might be a situation where a keyboard would need to be plugged to get that functionality.  Once I get to a point I can test this, I'll report what I find.

EnsignRutherford, 2024-01-08:
If you were into the Coco back then you must remember the work that Steve Bjork did, MegaBug, Sands of Egypt, just to name a couple of awesome titles.  I found out recently he passed, news from the Glendale Color Computer Club,

christianhaitian, 2024-01-08:
I remember the name but I never played those titles.  Hopefully this will be a chance to play them.

EnsignRutherford, 2024-01-08:
While I'm doing this I tried the Jedi Academy and Outcast ports.  A fix needs to be applied to the screen size.  Check out this reddit post for the specifics: https://www.reddit.com/r/RG353V/comments/yttkxg/port_master_jedi_academy_resolution_fix/

christianhaitian, 2024-01-08:
I no longer handle port master ports.  There's a separate team that handles those.

EnsignRutherford, 2024-01-08:
Okee... I'll send this to them.  Easy peasy fix.

EnsignRutherford, 2024-01-08:
Oops... coco_carts.xml: https://github.com/mamedev/mame/blob/master/hash/coco_cart.xml
and coco_flop.xml: https://github.com/mamedev/mame/blob/master/hash/coco_flop.xml

EnsignRutherford, 2024-01-09:
I just remembered that for my arcade build to play some ports without a keyboard I used either qjoypad for joy2key.  I will look into that.

christianhaitian, 2024-01-09:
We can use gptokeyb
https://github.com/christianhaitian/gptokeyb

EnsignRutherford, 2024-01-09:
I had a dog and his name was BINGO!

EnsignRutherford, 2024-01-09:
Leveraging love.sh as an example I now have it setup where a unique gptokeyb file is created for each game if not already defined and a default gptk is copied from /opt/coco3.  Testing out things now.

christianhaitian, 2024-01-09:
Cool.  You can also use `mvem.sh` as an example as well.

christianhaitian, 2024-01-09:
Ok, I finally got a chance to try this out and as usual, there's some hurdles that have to be cleared to play these systems when dealing with MESS/Mame.  For Canyon Climber, I had to make sure to have the coco3.zip bios file in the coco3 folder and it had to contain the `coco3.rom` and `disk11.rom` files.  I had to make sure the zip file was named cclimber.zip and within that file, it had to have `Canyon Climber (1982) (26-3089) (DataSoft) [a1].rom` in it.  I could only control the game with the joystick and action buttons but that might just be needing to redo the controller configuration.

Either way, it's definitely a process to run these games.  Thank goodness for the retrobat documentation [here](https://wiki.retrobat.org/systems-and-emulators/supported-game-systems/home-computer/trs-80-color-computer#content-of-bios-files) related to what needed to be in the respective bios zip files.

christianhaitian, 2024-01-09:
So cool to play Rampage on the CoCo 3 via emulation on this.  Thanks again for this suggestion and the research on getting this running.  For keyboard input, gptokeyb allowing custom keyboard entry will have to do.

christianhaitian, 2024-01-09:
Ok, looks like you can just run the .ccc files directly as long as their named as the short name shown in the xml file.  For instance, `Canyon Climber (1982) (26-3089) (DataSoft) [a1].ccc` needs to be renamed to `cclimber.ccc` and then it loads fine.

christianhaitian, 2024-01-09:
Good News, there's a workaround for the keyboard bind need for some games.  See [here](https://forums.launchbox-app.com/topic/61379-a-quick-hack-for-retroarch-cores-that-dont-have-a-keyboard-controls-option/).  Just load this remap file and you can assign a keyboard key to any button.  For instance, I assigned the start key the letter Y and I could start Rampage just fine.  Better than setting up a keydaemon for this.

[coco3.zip](https://github.com/christianhaitian/arkos/files/13881214/coco3.zip)

EnsignRutherford, 2024-01-10:
Let me check it out and I will update my configuration accordingly!

christianhaitian, 2024-01-10:
The one I attached in the message above your last post is what I used.  It has not keys assigned but the device will show as unknown and you can assign any keyboard key to it from the assignment menu for each key.

EnsignRutherford, 2024-01-10:
I noticed the same behavior.  On the third level of Canyon Climber the player walks off the edge if you are not fast enough to stop.  Interestingy some games generate segmentation faults, e.g. robocop, upon starting.  Not sure why,

EnsignRutherford, 2024-01-10:
> The one I attached in the message above your last post is what I used. It has not keys assigned but the device will show as unknown and you can assign any keyboard key to it from the assignment menu for each key.

Ok so I'm trying to figure out how I create a rom-specific remap file.  I now have a coco3.sh script that executes the rom.  It creates a custom retroarch.cfg which points the saves into a saves subfolder under coco3.

EnsignRutherford, 2024-01-10:
Ok I figured out where the filles are stored per content.  So my coco3.sh will copy a default coco3.rmp to that location in "/home/ark/.config/retroarch/config/remaps/MAME" for each one if it doesn't already exist.  I was trying  to move it to under the coco3 folder to have it autoload by setting input_mapping_path in the appended retroarch_coco3.cfg but that wasn't working.  Now have to figure out how to support mouse AND keyboard.  Almost there.

christianhaitian, 2024-01-10:
gptokeyb can do mouse emulation.  Not sure how it will work with retroarch though.

EnsignRutherford, 2024-01-10:
Ok.  Got the retroarch.cfg working now where it goes to a custom folder for the configs.  :)

EnsignRutherford, 2024-01-12:
Ok... :)  I have 76 cartridges that are able to run, each having it's own remap file so it can be configured to work with keyboard characters!   Predator and Robocop don't run because they are larger than 16K cartridges and the earlier versions of mess don't support that size so they segmentation fault.  Oh well.

Is there a way to compile xroar for the RockChip?  I had to make a custom rom/game set that natches the coco_carts.xml in order for them to run.

christianhaitian, 2024-01-12:
What is xroar?  Can you link the source?

EnsignRutherford, 2024-01-13:
https://www.6809.org.uk/xroar/.  It's standalone but supports a lot more than MESS or MAME do, at least the versions that are in ArkOS.

christianhaitian, 2024-01-14:
I was able to successfully build and run xroar.  It seems like it's easier to run game carts that are 8Kb in size or less.  I can't seem to get bigger carts to run though like Rampage, Robocop, or Castle of Tharoggad.  Attached is the binary and launch script `xroar.sh` in `/usr/local/bin`.  You can restore this backup that will install everything needed except for the coco3.rom and disk11.rom files needed for coco3 emulation.  The script is set to look for bios from the `/roms/bios` or `/roms2/bios` folder depending on where you launch your carts from.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, you should be able to launch your coco3 games by doing `xroar.sh _<path to rom>_` from a terminal session.  

Make sure the emulationstation has been shutdown before you attempt this so you can see how it works.  Hopefully you can help figure out how to run games larger than 8k.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/13930340/arkosbackup.tar.gz)

EnsignRutherford, 2024-01-14:
I'll start looking at this today.  Rainy day.

EnsignRutherford, 2024-01-14:
> I couldn’t get them to run in my windows install of xroar. Perhaps it’s the bios or game files I used? I was attempting to use the .ccc cartridges of those games. What all bios files did you install?

Predator I can play on Windows and the RG353V.  Robocop gets to the splash screen on both then hangs.  Castle does not run.  :(

christianhaitian, 2024-01-14:
I never tried Predator.  Robocop and castle i got the same results.  Rsmpage doesn’t run either.

christianhaitian, 2024-01-14:
I wonder of an older version will run them?  Maybe a regressive in the new version.

christianhaitian, 2024-01-14:
So far, xroar seems to run 8k cartridges and lower fine.  It's much easier to deal with this emulator vs MESS.  Maybe future versions will add the ability to run larger titles.

christianhaitian, 2024-01-14:
Even at the color computer archive, xroar there can't play larger roms either so it must be a limitation of the emulator.  I've emailed the developer to see if this is known and if there's any plans to eventually review the issue.

christianhaitian, 2024-01-14:
Until, then I'm thinking of including this emulator and making it the primary choice with MESS as another option since it can load more coco3 games but control is just wonky with that.

EnsignRutherford, 2024-01-14:
I was going to setup es_systems.cfg to support both.

christianhaitian, 2024-01-14:
Already did.

```
	<system>
	    <name>coco3</name>
	    <fullname>Tandy - Color Computer 3</fullname>
	    <path>/roms/coco3/</path>
	    <extension>.7z .7Z .bin .BIN .cas .CAS .ccc .CCC .dsk .DSK .rom .ROM .wav .WAV .zip .ZIP</extension>
	    <command>sudo perfmax %GOVERNOR%; nice -n -19 /usr/local/bin/coco.sh %EMULATOR% %CORE% %ROM%; sudo perfnorm</command>
	    <emulators>
	       <emulator name="standalone">
		   </emulator>
		   <emulator name="retroarch">
	         <cores>
			   <core>mess</core>
			 </cores>
		   </emulator>
		</emulators>
	    <platform>coco</platform>
	    <theme>coco3</theme>
</system>
```

contents of `coco.sh` to be placed in the `/usr/local/bin` folder
```
#!/bin/bash

if [[ $1 == "standalone" ]]; then
  # We'll get the root folder here so we can pass that info to the executable for located bios files
  directory=$(dirname "$2" | cut -d "/" -f2)

  # If there isn't an existing xroar keydemon, we'll create it here from another existing keydemon
  if [ ! -f "/usr/local/bin/xroarkeydemon.py" ]; then
    sudo cp -f /usr/local/bin/drastickeydemon.py /usr/local/bin/xroarkeydemon.py
    sudo sed -i 's/drastic/xroar/g' /usr/local/bin/xroarkeydemon.py
    sudo chmod 777 /usr/local/bin/xroarkeydemon.py
  fi

  # Allows for killing the emulator using select+start (minus + start for rgb10 and 1 + start for Chi) 
  sudo /usr/local/bin/xroarkeydemon.py &

  /opt/xroar/xroar -rompath /$directory/bios -machine coco3 -ram 2048K -run "$2"

  # Just in case the emulator exits without needing for the python3 keydemon, we'll kill it here
  sudo killall python3

  # Sometimes the global hotkey demon is negatively impacted by a python keydemon, let's restart to make sure it is ok
  sudo systemctl restart oga_events &
else
  /usr/local/bin/"$1" -L /home/ark/.config/"$1"/cores/"$2"_libretro.so "$3"
fi
```

christianhaitian, 2024-01-14:
platform has to be set to either `coco` or `tandy` so the games can be scraped via screenscraper

christianhaitian, 2024-01-16:
I thought of that as well.  My idea was if a zip file is detected and xroar is being launched, unzip the rom to a memory location `/dev/shm` and run the rom from there.  This was the rom won’t get added to the coco3 folder and potentially cause a duplicate entry.

christianhaitian, 2024-01-16:
This is assuming that the only file in the zip file is a singular .ccc file since I haven’t been able to autorun anything but a .ccc or .rom file with xroar yet.

EnsignRutherford, 2024-01-16:
Yeah I tried with some DSK files and it doesn't look like xroar supports it.  Though it is documented to autorun .BIN files, which will only allow some games to run.

EnsignRutherford, 2024-01-16:
We think a like.  :)   to a certain point if someone wants to use a
Bluetooth keyboard on the RG353V they can.  Most games can play though with
some caveats.

christianhaitian, 2024-01-17:
What I've come up with so far for supporting expansion of zip files for xroar.  Tests so far seems to work fine.
```
#!/bin/bash

if [[ $1 == "standalone" ]]; then
  # We'll get the root folder here so we can pass that info to the xroar executable for locating bios rom files
  directory=$(dirname "$2" | cut -d "/" -f2)
  # The next 2 variables are for use with custom game controls if the user creates some
  gamecontrols=$(echo "$(ls "$2" | cut -d "/" -f4 | cut -d "." -f1)")
  custom_gamecontrols_nocase=$(find "/$directory/coco3/controls" -maxdepth 1 -iname "${gamecontrols}".gptk)
  # The next variable is for getting the extension of the game file
  ext="${2##*.}"

  # The next 3 lines below are for preparing the environment so the keyboard/mouse emulator can successfully launch
  sudo chmod 666 /dev/tty1
  sudo chmod 666 /dev/uinput
  export SDL_GAMECONTROLLERCONFIG_FILE="/opt/inttools/gamecontrollerdb.txt"

  # This guard is here because there's a weird issue with the latest SDL2 that makes the odroid go advance type units controls look weird
  # So we'll use an available older sdl2 just for the keyboard/mouse emulator if need be.
  if [[ -e "/dev/input/by-path/platform-odroidgo2-joypad-event-joystick" ]]; then
    PRELOAD="LD_PRELOAD=\"/usr/lib/aarch64-linux-gnu/libSDL2-2.0.so.0.18.2\""
  else
    PRELOAD=""
  fi

  # Here we can load custom game controls instead of the default controls.
  # Users can create their own mappings for each game.  
  # Just create a controls subfolder within the roms/coco3 (roms2/coco3 for 2 sd card setups) 
  # folder and create a text file named exactly similar to game name but with a .gptk extension.
  # See https://raw.githubusercontent.com/christianhaitian/arkos/main/mvem%20pics/mvem.gptk
  # for an example of how to setup the structure of this file.
  # Unused gamepad keys should be commented out with \" like the start key is in the example .gptk
  # file linked in the previous sentence.
  # To enable keyboard mode for the emulator, insert #keyboard mode at the top of the file
  # To enable mouse mode, insert #mouse mode at the top of the file
  if [ -f "$custom_gamecontrols_nocase" ]; then
    echo "Loading custom user controls from $custom_gamecontrols_nocase" >> /dev/tty1
    /opt/inttools/gptokeyb -1 "xroar" -c "$custom_gamecontrols_nocase" &
    if [[ "$(cat "$custom_gamecontrols_nocase")" == *"#keyboard mode"* ]]; then
      params="-joy-left kjoy0"
    elif [[ "$(cat "$custom_gamecontrols_nocase")" == *"#mouse mode"* ]]; then
      params="-joy-left mjoy0"
    else
      params="-joy-left joy0"
    fi
  else
    echo "Loading default controls /opt/xroar/controls/xroar.gptk" >> /dev/tty1
    ${PRELOAD} /opt/inttools/gptokeyb -1 "xroar" -c "/opt/xroar/controls/xroar.gptk" &
    params="-joy-left joy0"
  fi

  # The standalone xroar emulator doesn't support zip files.  We'll take care of that here
  if [[ "${ext,,}" == "zip" ]]; then
    if [ ! -d "/dev/shm/cocoroms" ]; then
      mkdir -p /dev/shm/cocoroms
    else
      rm -rf /dev/shm/cocoroms/*
    fi
    unzip -qq -o "$2" -d /dev/shm/cocoroms/
    for CART in ccc rom bin; do
      GAME=`find /dev/shm/cocoroms/ -iname "*.${CART}" | head -n 1`
      if [ ! -z "$GAME" ]; then
        break;
      else
        printf "\nCouldn't find a compatible cartridge or type .ccc or .rom from $2\n" > /dev/tty1
        sleep 5
        exit
      fi
    done
  else
    # No file with a .zip extension is detected so we can just pass the file directly to xroar
    GAME="$2"
  fi

  # We're now ready to try and run the game
  /opt/xroar/xroar -rompath /$directory/bios -machine coco3 -ram 2048K -run "$GAME" ${params}

  # Done running the game.  Time to clean up the temp memory space if needed.
  if [ -d "/dev/shm/cocoroms" ]; then
    rm -rf /dev/shm/cocoroms/*
  fi

  # Let's unset the SDL Gamecontroller config file in the environment and make to to kill gptokeyb if it's still running somehow.
  unset SDL_GAMECONTROLLERCONFIG_FILE
  if [[ ! -z $(pidof gptokeyb) ]]; then
    sudo kill -9 $(pidof gptokeyb)
  fi

  # Make sure the hotkey daemon is still fine, clean up the screen buffer, then we outta here!
  sudo systemctl restart oga_events &
  printf "\033c" >> /dev/tty1
  exit 0
else
  # Just run this game with libretro mess instead.  Good luck!
  /usr/local/bin/"$1" -L /home/ark/.config/"$1"/cores/"$2"_libretro.so "$3"
fi
```

EnsignRutherford, 2024-01-17:
Ahhh, your code is somewhat different then mine.  If I didn't find a custom game control file I created a default one for it.  And I just overwrote my code.  :(

christianhaitian, 2024-01-17:
Why create a default if one is not provided by the user?  Games like Canyon Climber don’t need it.

christianhaitian, 2024-01-17:
Oh gotcha.  i forgot the control mess of MESS!

EnsignRutherford, 2024-01-17:
Since you can configure MESS more easily I was making it the default emulator and only using xroar for the 'difficult' games or games that are on DSK, Cassette, etc.  Though I haven't totally gotten that figure out yet.

EnsignRutherford, 2024-01-17:
This is the ode I re-wrote.  Have to test it later.  This is for the if not standalone:

```
else
  # Get the ROM folder to use for setting up the control files
  directory=$(dirname "$3" | cut -d "/" -f2)

  # Get the game name to use for the controls file
  gamecontrols=$(echo "$(ls "$3" | cut -d "/" -f4 | cut -d "." -f1)")

  # Repoint the mapping directory in an appended config file
  if [[ ! -d "/$directory/coco3/cfg" ]]; then
    mkdir /$directory/coco3/cfg
  fi
  echo "input_mapping_directory = "\"/$directory/coco3/controls\" > /$directory/coco3/cfg/retroarch_coco3.cfg

  if [[ ! -d "/$directory/coco3/controls" ]]; then
    mkdir /$directory/coco3/controls
    mkdir /$directory/coco3/controls/MAME
  fi
  cp /opt/coco3/coco3.rmp /$directory/coco3/controls/MAME/$gamecontrols.rmp

  # Just run this game with libretro mess instead.  Good luck!
  /usr/local/bin/"$1" -L /home/ark/.config/"$1"/cores/"$2"_libretro.so --appendconfig=/$directory/coco3/cfg/retroarch_coco3.cfg "$3"
fi
```

From my tests this creates an 'empty' remap folder with nothing reassigned so existing controls that work will work, e.g. Canyon Climber.  I'll test this tonight as I just wrote it without testing as I'm away from the game console.

christianhaitian, 2024-01-18:
Very cool.  We can definitely automate that.

EnsignRutherford, 2024-01-18:
[coco.zip](https://github.com/christianhaitian/arkos/files/13980570/coco.zip)

[81 more replies]

closed 2024-01-27 by christianhaitian