christianhaitian/arkos #896: Standalone emulators and other cores for n64 broken in ArkOS after update) last update (R36S)

Guzelota, 2024-01-08:
All N64 cores (except the default parallel) have stopped working. After attempting to launch, the screen goes black and I am back in my N64 gamelist. In retroarch you can't even load the core. 

I quickly reinstalled everything on a second SD card (with the RG351M-Arkos image) and installed the update again (via the online function in the system options) - then it didn't work again. 

For final clarification, I reinstalled everything on an SD card, but then blocked the N64 cores for updates using the retroarch settings. After the system online update, the games continue to start regularly via Mupen - so it must be the core / some config things that the Arkos update breaks.

Guzelota, 2024-01-08:
(I am not the only one with this problem. There are some new reddit threads with the same problem:

https://www.reddit.com/r/SBCGaming/s/KHmymXs33t

https://www.reddit.com/r/SBCGaming/s/OZcjf5V0iD

christianhaitian, 2024-01-08:
This is due to the way the stock ArkOS image provided with the R36s provisions some items which causes the update to not place the right files .  Updating from the RG351MP image should have worked though so not sure what happened there.  Try restoring the following file.  This should resolve the N64 standalone issues.  

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, you should be able to launch your n64 games using the standalone emulators.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/13855510/arkosbackup.tar.gz)

Also note, a community member by the name of Aeolus has an image specifically built for the R35s and R36s units with these fixes in place for those units.  The links for those images are available here:

ArkOS_R35S-R36S_v2.0_12222023 (By Aeolus)
https://drive.google.com/file/d/10z7j7IZ7WX3y10ZJBW_a2-agcIe1Dx9m/view?usp=drive_link
https://www.mediafire.com/file/da4193t5vsjfsbr/ArkOS_R35S-R36S_v2.0_12222023.img.xz/file

More information about this image and further discussion and further developments related to the R35s/R36s units are available on the Retro Handheld Discord.  https://discord.gg/retrohandhelds

Guzelota, 2024-01-08:
Thank you for the quick reply and your solution to the problem. I assume that this "restore-backup" method could also be used for other cores (i.e. also the mupen64plus-next) if the corresponding folder paths are taken into account? Unfortunately, this also causes Retroarch to crash (also only since the update). 

When I look in the corresponding folders, Mupen seems to have spread quite widely on my system anyway. :D

[image]

[image]

christianhaitian, 2024-01-08:
The contents of that mupen64plus folder indicates to me that the update didn't know what to do with that device.  Best to to load that community members 's updated ArkOS image that I linked above.

mrglennjones, 2024-01-08:
> This is due to the way the stock ArkOS image provided with the R36s provisions some items which causes the update to not place the right files . Updating from the RG351MP image should have worked [...]

hi cristian, im having an issue writing these 2 files (erroring at 90%), tried etcher and win32diskmanager.

christianhaitian, 2024-01-08:
Head to the gameconsole-r35s-r36s channel in the Retro Handhelds Discord for assistance with that image.  It's not one I maintain or support.

DK-1980, 2024-02-21:
> This is due to the way the stock ArkOS image provided with the R36s provisions some items which causes the update to not place the right files . Updating from the RG351MP image should have worked [...]

Hi,

Thank you for the support. I am new in the handheld emulation world so sorry if my question sounds dumb. After updating the ArchOS for my r36s; it stopped starting standalone emulators; so would this method also work for psx and dreamcast? Thank you!

christianhaitian, 2024-02-21:
Yes.  Also, don't use the sd cards that came with the unit.  They will fail quickly.  Source your own good quality SD cards that are known to work with the r36s.  The unit is sensitive to sd cards so refer to the compatibility list [here](https://docs.google.com/spreadsheets/d/1gWxtr-GmwWop-_qGUq022RXxK2aTLpPg9Qra68TQLI8/edit#gid=0) for known good working SD cards for the unit.

DK-1980, 2024-02-22:
> Yes. Also, don't use the sd cards that came with the unit. They will fail quickly. Source your own good quality SD cards that are known to work with the r36s. The unit is sensitive to sd cards so [...]

Thank you so much!