christianhaitian/arkos #900: RG353M HDMI out works until a game is started

Asuza, 2024-01-09:
2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki? **Yes**

4. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions? **No, it applies to multiple cores**

5. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ? **Yes**

### Describe the Issue (If applicable)
When using HDMI out, the ArkOS bootup and EmulationStation work great. However, once a game is selected, the HDMI signal is lost. I believe the device may be freezing because pressing Select+Start does not go back to EmulationStation. I have had to use the Reset button to get back to EmulationStation.

### How can the issue be reproduced? (If applicable)
1. Connect RG353M to a screen using HDMI
2. Start a game (Super Nintendo, Game Boy Advance, and Capcom Play System II were tested)
3. HDMI signal is lost, seemingly until a restart

### What device are you using?
RG353M

### What version of ArkOS are you on?
12222023

### Anything else you'd like to include that may help to review this issue or feature request?
If there are any kind of debug logs that would help, I can SSH into the device to get them. I just need to know where to look.

christianhaitian, 2024-01-10:
See here https://github.com/christianhaitian/arkos/issues/828

closed 2024-01-10 by christianhaitian

Asuza, 2024-01-10:
Just some notes, though I still don't know what combinations of things are a problem. I only have [one micro HDMI adapter](https://www.amazon.com/gp/product/B09MD23K4X), which was used in all cases.
- Two different HDMI cables (one from a Nintendo Switch) would show EmulationStation on a Dell monitor, but games would not appear.
- One slightly older HDMI cable from a different Nintendo Switch worked fully on a very old Vizio TV.

I intend to add more notes and details when I have more time.