christianhaitian/arkos #901: Display reset loop when running M8c

Bigpeeet, 2024-01-11:
Hi Christian,
Thanks for your work on the recent update. I'm guessing subfolder support is now active again?

I'm still having issues running M8 headless on my RG351V. I've been using file manager to execute the setup files in subfolders and run the M8 executable (won't run from the regular GUI) and have had success with it starting up, but I'm getting a screen reset loop happening. When I run the executable file from ssh it gives me the following on repeat:

ERROR: Invalid packet
INFO: Reset display

Every so often, the ERROR lists this as further information

ERROR: Invalid draw character packet: expected length 12, got 7

Apologies if this should be with the developer of M8C, I'm not a programmer and really not sure who can assist best with this. Do you have any ideas about how to resolve this so the display stops glitching?

Many thanks in advance

christianhaitian, 2024-01-11:
Have you tried enabling `scan ports folder on boot` from the start>UI Settings section and running it directly from the emulationstation menu?

Bigpeeet, 2024-01-11:
Yes, that's enabled at the moment

christianhaitian, 2024-01-11:
Then I'm out of ideas of what the issue could be.  It could be it has an issue with the current sdl2 graphic library.  Maybe a solution like this here (https://github.com/christianhaitian/arkos/issues/777#issuecomment-1693766791) may help.  Otherwise, better check with the developer of m8c.

closed 2024-01-11 by christianhaitian