christianhaitian/arkos #904: Feature Request - enable fn button on R36S

ekoprihastomo, 2024-01-13:
Just want to let you know that someone made fn button on R36S functional now
Follow below link to get the files needed...
[https://github.com/AeolusUX/R36S-DTB](url)

christianhaitian, 2024-01-13:
I don't support the R36S as I don't have a unit nor will I add official support for the unit as I'm not looking to add support for any additional units at this time.  I will look to see if this change can be integrated into the rg351mp image without impacting the rg351mp.

closed 2024-01-13 by christianhaitian

christianhaitian, 2024-01-13:
For those with R35S units with the stock ArkOS firmware or those with R35S or R36S using the RG351MP ArkOS firmware, the attached zip contains 4 files will add the ability to make use of the FN button on your units.  Just copy them to the fat32 boot partition on the OS card overwriting the existing files there.  You can then make use of hte button in emulators like Retroarch.  Thanks to @Aeolus and Hack Around for the guidance on enabling this.  These files will also be included in the next ArkOS update later this month.

[R35s.zip](https://github.com/christianhaitian/arkos/files/13929472/R35s.zip)

ekoprihastomo, 2024-01-13:
Thank for the reply 🙏
I already have it working with my R36S, I've tested it and it's working, just want to tell you in case you don't know it yet
My request is more like request for people who don't tinker with their device rather than my personal request
Large portion of people don't upgrade and most people who do upgrade they upgrade via flashing or OTA, only a handful of people will bother with replacing files from different source

ArkOS is an awesome OS and R36S have full feature board with unmatched price, just want people to get most out of their device
Did you know that R36S board support rumble motor? some R36S shipped with rumble motor installed but sadly most of them aren't
Even if you didn't get R36S with rumble motor installed you can mod it in yourself with minimal effort coz the board already support rumble and have empty space for it

BTW thank you soooo much for your work with ArkOS 🙇‍♂️
You bring smile to lots of people around the world 💯🎉