christianhaitian/arkos #905: A/B button swap

lauren7ino, 2024-01-16:
Hi Christian,

Context: I was an Xbox 360 user a long time ago and never got used to the A/B buttons being swapped in my RG353M, so after struggling to get used to the new layout for months, I gave up and decided to activate the A/B button swap settings.

On EmulationStation: Main Menu > Advanced Settings > Switch A/B Buttons in EmulationStation
and on in both RetroArches: Settings > Input > Menu Controls > Menu Swap OK and Cancel buttons
Now everything's great and my old brain no longer short circuits.

There is still one area where these settings don't apply:
The tools on the Options menu and the popup keyboard. For example the Update tool presses the buttons with the A button still.

Do you know a way to fix these?
Thank you in advance

christianhaitian, 2024-01-16:
Hi,

No.  That feature only swaps it for the ES menu and nothing more.  Not something I will focus on supporting in the near future either.

closed 2024-01-16 by christianhaitian

lauren7ino, 2024-01-16:
OK, thanks.

lauren7ino, 2024-01-26:
Hi again,
Your tutorial (below) can be used to swap A/B buttons in the full system at once?

https://github.com/christianhaitian/arkos/wiki/Key-codes-and-Global-Hotkey-config-file-locations#rg353mvvs-and-rg503

If I set:
A = 304
B = 305
?

christianhaitian, 2024-01-26:
That would only work for some SDL applications that make use of gamecontrollerdb.txt.  The easiest solution would be to build a new dts file that swaps those buttons system wide.

Somewhere around this line https://github.com/christianhaitian/RG353VKernel/blob/6c6eba2917cdcb35c4f1d6d558367c481a893f75/arch/arm64/boot/dts/rockchip/rk3566-353m.dts#L431

and this line https://github.com/christianhaitian/RG353VKernel/blob/6c6eba2917cdcb35c4f1d6d558367c481a893f75/arch/arm64/boot/dts/rockchip/rk3566-353m.dts#L426

lauren7ino, 2024-01-26:
Wow, applying these settings would be wonderful !
I know hot to branch a project, but unfortunately I don't have the time to invest learning how to recompile and publish a core to the device (I don't have much expertise in Linux platform). My Linux is in a VM ARM64 (I own an Apple Silicon Mac). I don't even know if it compiles properly.
Thank you for the post. I't good to know it can be done.

lauren7ino, 2024-04-06:
Hi
Returning to this subject, can you give me some pointers on how I start?
I have a woking Ubuntu ARMx64
1. Branch the project
2. Modify the files
3. Run DTS makefile to create a new DTB
4. Replace the existing boot\rk3566-OC.dtb and rk3566-OC.dtb.bright
Is this it?
Thanks