christianhaitian/arkos #907: Updater not working / settings not saving

uniQ168, 2024-01-18:
I'm having issues using my RG353M with ArkOS 2.0 (08112023).
Changes in the options menu are not saved and the updater does not work.
I get the following error message: "There was an error with attempting this update. Did you make sure to enable your wifi and connect to a wifi network? If so, enable remote services in options and try to update again".

Well I am properly connected to my wifi and even after enabling the remote services, I could not update. So I connected via SSH and ran the updater from there, resulting in the following output:
"Checking for updates.  Please wait...mode of '/home/ark/ArkOSUpdate.sh' retained as 0777 (rwxrwxrwx)
tee: /home/ark/esupdate.log: No space left on device
librga:RGA_GET_VERSION:3.02,3.020000
ctx=0x55953e8eb0,ctx->rgaFd=3
Rga built version:version:+2017-09-28 10:12:42
obj=0x55782dffa8, event=12"

Checking my devices, I get:
ark@rg353m:/opt/system$ df
Filesystem     1K-blocks      Used Available Use% Mounted on
udev              995552         0    995552   0% /dev
tmpfs             201684      2316    199368   2% /run
/dev/mmcblk1p4   7255536   6959732         0 100% /
tmpfs            1008408         4   1008404   1% /dev/shm
tmpfs               5120         4      5116   1% /run/lock
tmpfs            1008408         0   1008408   0% /sys/fs/cgroup
/dev/mmcblk1p3    100784     32237     68547  32% /boot
/dev/mmcblk1p5 242737264 185159584  57577680  77% /roms
tmpfs             201680         0    201680   0% /run/user/1002

Anyone else had this problem? Is there a solution?

christianhaitian, 2024-01-18:
`/dev/mmcblk1p4 7255536 6959732 0 100% /`
Your sd card is either to small or you've been installing many applications on the linux partition of the sd card and you've run out of space.

uniQ168, 2024-01-18:
I have a 256G SD Card, flashed ArkOS on it and copied over roms. I don't know how this partiton even filled up to 100%.
Any suggestion on how to fix this? The sd card hast plenty of storage left.

uniQ168, 2024-01-19:
Pretty sure I followed the documented standard installation, but I'm not entirely sure to be honest.
I really want to avoid a complete reinstall/reconfiguring and just expand the partition if it's possible. :/

christianhaitian, 2024-01-19:
Check if there’s a `/tempthemes` folder on the sd card.  If so, that means you did not insert the sd card into the unit first and allow the initialization.  Which would also mean you used a tool like minitool partition wizard to creat and/or expand the EASYROMS partition before the initial boot.

uniQ168, 2024-01-19:
I just checked and there is no such folder.

christianhaitian, 2024-01-19:
Ok.   Then unless you've been in there apt installing other programs, I'm not sure why the root partition is full.  You should have at least 2GBs free there.  What else did you do with the system after loading it on to your 256GB card and loading your games into the ROMs partition?

uniQ168, 2024-01-20:
Literally nothing. I just copied over my roms, set up RetroArch and played some games.

christianhaitian, 2024-01-20:
Unfortunately, you'll have to reflash the OS to fix the issue.

uniQ168, 2024-01-22:
Reflashed and reconfigured my device. I have 2,4G space left on /dev/mmcblk1p4 and will keep an eye on it.
Thanks for your support.

closed 2024-01-22 by uniQ168

christianhaitian, 2024-01-22:
Strange it happened with the initial flash but happy it’s resolved now👍🏿