christianhaitian/arkos #909: EmualtionStation game delete

lauren7ino, 2024-01-18:
Hi
I know this is probably a duplicate question, but I must ask:

Background:
I have hundreds of games on my card, I automatically selected (program I created) by keeping the ones with the highest rating in screenscraper. All very nice and it's a great collection I'm proud of.
All have great ratings, but unfortunately not all games are up to my taste.

Q: How do people manage (mostly hide) games that don't like?
I mean, without leaving EmulationStation?

Thanks

christianhaitian, 2024-01-18:
Some options below with number 1 making the most sense imo.

1.  Remove the games you don't want from your sd card.  Why keep games you don't want in the first place taking up needless spaces on your sd card and further slowing the initial boot into the OS even if it's just a few milliseconds?
2.  Favorite the games you plan to play so that they can be included in the Favorites gamelist and you can just go there to pick one of your favorited games.
3.  If you feel compelled to keep all of these unwanted games on your sd card, you can hide each of them by hitting Select on the games then going to Edit This Game's metadata, and mark the game as hidden there.

closed 2024-01-18 by christianhaitian

lauren7ino, 2024-01-23:
Ha! I missed the Edit of Game's metadata. Thanks! Hidden is what I was searching for.