christianhaitian/arkos #911: Game PSP Loco Roco 2 make my R36S freeze and resetting ppsspp settings

Midas086, 2024-01-20:
I use R36S with stock ArkOS v2.0 (12/22/2023)
Today, I tested Loco Roco 2 on my r36s and as you can see in the video, it caused my device to freeze. Can't do anything other than press the reset button.
After rebooting I tried again to make sure this was a serious error. and yes, this error causes ppsspp emulator to reset all settings including key mapping. The round, square, triangle and cross buttons no longer match the original settings, they are all turned upside down. Even the L3 button to load save state no longer works. I was quite confused and was planning to re-flash the rom, but luckily I found a way. That is to copy the default SYSTEM folder at the path PSP > PPSSPP > PSP on the OS memory card and paste it into the memory card containing the game also following the path PSP > PPSSPP > PSP. Restart your computer and everything will be back to how it was. Going to options, advanced settings, and restore default controls for ppsspp doesn't help in this case because it still loses r36s default key mapping settings.
Even though I have mentioned how to fix it, if you encounter this error again, all ppsspp settings will be reset.
[image]

christianhaitian, 2024-01-20:
Sounds like an issue with your game.  have you tried obtaining another copy of this game and trying that?

closed 2024-01-20 by christianhaitian

Midas086, 2024-01-20:
I tried another version. The first time it was the EUR iso file, it caused an error so I thought maybe the game file was corrupted so I redownloaded the same game but the USA version. The result is still an error. But that does not matter. If it's just a buggy game then I'll simply ignore it and not play it. But the problem I want to report here is that this crash error causes the computer to freeze completely and not be able to do anything. and more importantly, immediately after reboot ppsspp's settings are returned to default. This led to the fact that when playing another game, I could not use the L and R buttons, the square and triangle buttons corresponding to the X and Y buttons on the actual keyboard were reversed. The same goes for round and cross buttons. This is clearly a bug.

christianhaitian, 2024-01-21:
Ok.  Nothing I will fix for R36s as I don't support that device and I can't replicate the issue on my rg351mp unit.

christianhaitian, 2024-01-21:
If you're using the rg351mp image, there's a Restore Default PPSSPP Controls script available in the Options>Advanced section.  Good luck.

Midas086, 2024-01-21:
i guess my rom is based on rg351mp so when i go to Preferences>Advanced section>Restore default PPSSPP controls it doesn't or dynamic. it still has the same error. Perhaps the layout or key names of the rg351mp are different from the r36s. Anyway, thank you very much!