christianhaitian/arkos #916: Auto-sleep?

lauren7ino, 2024-01-23:
Hi,
I know this is uncommon, but just happened to me:
I initiated a full scrape of the game's images and videos, something came up, I placed the console in the shelf and totally forgot about it.
I believe between the scraping and video-screensaver, the battery went zero (was 99% when it started).
Are there any plans to add a auto-slepp/auto-shutdown timer?
Thanks

lauren7ino, 2024-01-23:
Update: My situation is grim... it seems it is not charging besides the orange led being on. It won't power on, not even with the adapter.
(RG353M)

christianhaitian, 2024-01-23:
No plan for auto sleep for any supported devices at this time.  Try holding the reset button for 20 seconds to see if that revives the unit.  If not, then you'll have to open the device and unplug the battery for 20 seconds, then plug it back in.  This last step normally resolves the issue.

closed 2024-01-23 by christianhaitian

lauren7ino, 2024-01-24:
Thanks. Manage it fix it by letting it charge all night long on a powerbank.

I thought the auto-shutdown was a easy thing to add. Like running a script like so:
https://unix.stackexchange.com/questions/84437/how-do-i-make-my-laptop-sleep-when-it-reaches-some-low-battery-threshold