christianhaitian/arkos #921: Sound from speaker suddenly does not work on Anbernic RG353VS

Domingster, 2024-01-26:
1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

Finding setting of sound and nothing found

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

### How can the issue be reproduced? (If applicable)

RG353VS

### What version of ArkOS are you on?

22122023

### Anything else you'd like to include that may help to review this issue or feature request?

Hello, suddenly sound from speaker does not work. System volume in Main Menu is 0% even if I try to increase it does nothing. + and - buttons of sound does work because when I hold F + these, it controls brightness. I am very confused because it worked properly and now it is not. Headphones works. Please can you help me what to try?

Domingster, 2024-01-26:
So after this post I did find this #862 and that was it, thank you!

closed 2024-01-26 by christianhaitian

Domingster, 2024-01-26:
And could you tell me how is this happening? It happened to me today second time after I charged device up. Is possible to repair this in next update? Thank you for response.

christianhaitian, 2024-01-26:
I've only had that happened to me once and a reboot normally fixes it for me.  I have no clue why some have this issue.

lauren7ino, 2024-02-02:
I had several problems with no audio on speakers, while I was using bluetooth AirPods while commuting, and speakers at night.
Several times I had to insert/remove a wired headphone to restore the speaker's audio.
Reboots won't fix the problem either.
There is definitely some bug related to auto-switching of audio outputs.