christianhaitian/arkos #923: Stuck booting on RG353VS in new v2.0 (01/27/2024)

Domingster, 2024-01-27:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

Yes

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

No

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

Yes

### How can the issue be reproduced? (If applicable)

-

### What device are you using? (Only the 

RG353VS

### What version of ArkOS are you on?

v2.0 (01/27/2024) 

When I firstly downloaded ArkOS I had problem with booting, everytime handheld stucks on ArkOS logo and never get further. After asking on reddit, some guy told me that I can try some edited file in BOOT folder which reduces 1,8 GHz to 1,4 GHz. It worked really well I was happy. It was on some version from this pages I think from October I am not sure, I used function Update in Options menu and I did update to version from 22.12.2023 and perfectly worked itself. But today I did Update to latest FW and it again stucks on booting screen. Fortunately fix file works again but I want to give you hint that 22/12/2023 there was no need to use this fix, so maybe there was something other which helps. Maybe you could offer here to flash SD card that version?

What do you think?

Thank you.

christianhaitian, 2024-01-27:
Hi,  this update included a new dtb file so not surprised you had to recopy that file.  That solution is one I won’t support as it reduces the max performance of the unit which I don’t want. I rather fix the actual problem if I ever  get a unit that has the issue so that I can review and resolve.

closed 2024-01-27 by christianhaitian

Domingster, 2024-01-27:
Thank you very much for response, Yeah it would be awesome, so hope sometime there will be anybody from your surroundings with this device… Have a nice weekend.

Domingster, 2024-02-08:
Maybe little idea, what about inspiring from JELOS creators? Their system boots without problems even in new updated versions, somaybe they got through this problem.

christianhaitian, 2024-02-08:
Already using the same uboot and kernel but there are still problems booting.