christianhaitian/arkos #924: Switch to SD2 For Roms on ArkOS 2.0 (1272024)

AeolusUX, 2024-01-27:
### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
After the update using Switch to SD2 For Roms doesn't work anymore.
It always says "Could not find a Fat, Fat32, Exfat, EXT4, or NTFS based sdcard to mount from slot2". This usually happens on unsupported TF Cards before but I have tested three previously supported TF Cards and they all have the same issue. Moreover I have manually edited the es_systems.cfg file and pointed them manually to roms2 and it still doesn't detect TF2. Switching back to the December ArkOS Build makes the cards work again.

### How can the issue be reproduced? (If applicable)
Freshly updated ArkOS > Switch to SD2 for Roms

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)
Game Console R35S

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
ArkOS 2.0 (1272024)

### Anything else you'd like to include that may help to review this issue or feature request?

Graviter, 2024-01-27:
same problem on r36s. If you update from a previous version where the second card for roms is already selected, it freezes at loading after the update. If you update a new system and then switch to the second card, it does not see the partitions on it.

christianhaitian, 2024-01-27:
What format are you using for your sd2 card?

Graviter, 2024-01-27:
exfat, clean install and update from 12222023 version

AeolusUX, 2024-01-27:
> What format are you using for your sd2 card?

I'm on exFat. I shall try other formats give me a sec.

AeolusUX, 2024-01-27:
NTFS didn't work either, so does LargeFat32.

christianhaitian, 2024-01-27:
I'll see if I can replicate this on my end with the RG351MP image on my R35s

christianhaitian, 2024-01-27:
I just pushed out an update that resolves the switch to sd2 card issue.  I don't provide support for the R36s unit but I suspect if you were on the previous stock OS, the card was just bad on that as the update would not impact an existing 2 sd card setup that was already active before the 01272024 update.

closed 2024-01-27 by christianhaitian

devrosx, 2024-01-27:
Hello looks like i have same problem or RG353VS, can i get also update for this platform

christianhaitian, 2024-01-27:
Not an issue on any supported rk3566 device including the RG353V.  I confirmed that I was able to switch to sd2 without issue.  Folders populated fine on my end on the latest update.  The issue is local on your end.

devrosx, 2024-01-27:
ok thanks for check after clean reinstall, bug was still same... but now i realized that problem was SD card format tried osx card formated with FAT, EXTFAT, FUSE-EXT2.. with all these it wasnt working (card was empty after switch to seconds sd card)...Then i tried to format my 256 GB card with FAT32 format (function on Raspberry Pi Imager osx APP) and then it worked... weird but now it works... thanks for your work :)