christianhaitian/arkos #925: Exiting PPSSPP using SELECT+START does not work on RG503

EnsignRutherford, 2024-01-28:
The ppssppkeydemon.py script assumes /dev/input/event4 is the retrogame_joypad.  That is true for the RG353V but not for the RG503, where it is /dev/input/event3.  The attached file finds the correct device regardless of the handheld and sets it up properly.
[ppssppkeydemon.zip](https://github.com/christianhaitian/arkos/files/14077964/ppssppkeydemon.zip)

EnsignRutherford, 2024-01-28:
I took a look at the other key demon python scripts, e.g. pico8keydemon.py and it assumes event3 or event4.  I cleaned up the code of the previous file with some additional error handling and it makes no assumptions, just the location of the devices inventory file.
[ppssppkeydemon.zip](https://github.com/christianhaitian/arkos/files/14078060/ppssppkeydemon.zip)

christianhaitian, 2024-01-28:
Interesting.  That keydemon wasn't properly updated from the last update so that's what caused that issue.  Your solution does seem more universal so i'll test it on the rest of the devices that ArkOS supports.  In the meantime, I'm placing this backup file here for others who run into this issue on a RK3566 device:

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, you should be able to use select+start to exit the ppsspp standalone emulators.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/14078324/arkosbackup.tar.gz)

EnsignRutherford, 2024-01-29:
I'm going to make the get_device function more generic so it can search for any.  I took a look at the other files and I see that they need access to power and volume.  I'll make one function that can be called from anywhere to do that.  :)

EnsignRutherford, 2024-01-29:
Did PortMaster get removed from the last build?  I had a bunch of ports installed and now none of them run and the Tools menu is missing from Options.

christianhaitian, 2024-01-29:
Nope.  Verify that portmaster files are in the tools folder within your roms folder

EnsignRutherford, 2024-01-29:
Weird.   The /roms/ports folder is empty on both RG353V and RG503 after the update.

EnsignRutherford, 2024-01-29:
I think the problem may be that there are two entries in fstab for the roms2 folder:
/roms2/tools /opt/system/Tools none bind
/dev/mmcblk2p1 /roms2 vfat umask=0000,iocharset=utf8,noatime,uid=1002,gid=1002 0 0

christianhaitian, 2024-01-29:
Your fstab is fine. That’s not your issue.  Weird as I haven’t experienced this on any of my units.  The update doesn’t touch the tools folder.  .  /opt/ToolReinstall portmaster per the instructions here 
http://portmaster.games/

EnsignRutherford, 2024-01-29:
> Your fstab is fine. That’s not your issue. Weird as I haven’t experienced this on any of my units. The update doesn’t touch the tools folder. . /opt/ToolReinstall portmaster per the instructions here http://portmaster.games/

Actually it's not...

/dev/mmcblk1p3 /boot vfat defaults,noatime 0 0
/dev/mmcblk1p5 /roms exfat defaults,auto,umask=000,uid=1002,gid=1002,noatime 0 0
#/roms2/tools /opt/system/Tools none bind
/dev/mmcblk2p1 /roms2 vfat umask=0000,iocharset=utf8,noatime,uid=1002,gid=1002 0 0
/roms2/tools /opt/system/Tools none bind

I commented out the third line and repeated it at the bottom.  Now the Tools menu is back in Options and the ports run again.
Because it appeared before the roms2 mapping it never got mapped.

christianhaitian, 2024-01-29:
Actually, your fstab is different than mine.

[image]

christianhaitian, 2024-01-29:
ignore mmcblk0 as that would be mmckbl1 for an sd card install.  The rest should look like that though unless you’ve been modifying your fstab manually?

EnsignRutherford, 2024-01-29:
I only did it now to fix that issue.  /roms2/tools was not being mounted on my build until I moved it to the bottom of the fstab file.

christianhaitian, 2024-01-29:
weird indeed.

christianhaitian, 2024-01-29:
No thanks as I just mostly use the RGB30 for tate games

EnsignRutherford, 2024-02-11:
Sorry for the delay.  Here is the rewritten ppssppkeydemon.py where the code is more generic per device.
[ppssppkeydemon.zip](https://github.com/christianhaitian/arkos/files/14232415/ppssppkeydemon.zip)

closed 2024-02-24 by christianhaitian