christianhaitian/arkos #926: Standalone-rice, standalone glide mk2 or gliden emulators immediately crash back to game list

adotagarwal, 2024-01-29:
I recently updated from the 2022-04 release to the most current 2024-01 release of ArkOS by running the update command under options.

After updating, i find that N64 games configured to use rice or glide* emulators immediately crash back to the game list (ES) where they used to open and work well.

I tried tracing down the logs to see what was going on, but only found an issue related to the audio, which I think is a red herring.

The issue can be reproduced by updating to the latest january image, setting an n64 game to use rice or glide, and then launching it.

I see the system logo, then a blinking cursor, then the system logo again, before it finally goes back to the N64 game list.

Are there any logs I can check to see whats going on? I enabled the debug logging and am looking at the log in /.emulationstation/es_log.txt -- this is where I saw the 'VolumeControl::init() - Failed to find mixer elements!' errors. I got rid of those by setting AudioDevice to Playback instead of Speaker but this didn't change rice/glide not opening

christianhaitian, 2024-01-29:
What device is this?

adotagarwal, 2024-01-29:
An r36s

christianhaitian, 2024-01-29:
I don't support the r36s.  This issue occurs with their stock firmware because of how they provisioned some key files used for detecting the version of a unit during updates.  Fixing this specific issue is simple though.  You can try restoring the attached backup file which should resolve that issue.  

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed, the N64 standalone emulators should work.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/14086420/arkosbackup.tar.gz)

However, there may be other issues caused by the update.  You may be better off using the AeolusUX ArkOS image available through one of the links below:

Google Drive
https://drive.google.com/file/d/10z7j7IZ7WX3y10ZJBW_a2-agcIe1Dx9m/view?usp=drive_link

Mediafire
https://www.mediafire.com/file/da4193t5vsjfsbr/ArkOS_R35S-R36S_v2.0_1272024-1.img.xz/file 

Good luck.

closed 2024-01-29 by christianhaitian

adotagarwal, 2024-01-29:
wow thanks!

1.  ill try the restore settings (i made a full image of the stock os, as well as the stock OS after i had finished setting up my device and after updating to 2024-01-27)-- so i can keep trying over and over again. I might diff the contents of the configs in that tar with the ones on my machine to learn more about which setting had changed.

2. as far as i can tell, everything else works much better than the original version. even psp performance is improved (if i use the standalone 2021 core)

3. i began this update journey because i couldn't get gta3 or vc to work in portmaster using the 2022/04 image

i had not heard about the aeolusUX image, but am downloading that too. Would you recommend that over the image you post here for rg351mp and the dtb file for the r35s? 

Lastly, are the r35s and the r36s not the same? I see that you fixed the FN button on the r35s in the latest changelog

christianhaitian, 2024-01-29:
aeolusUX image has all the updates including the FN key change for the R35s and R36s.  It's similar to the RG351mp image just without needing to swap any files or do any ota updates for the r35s/r36s units.

christianhaitian, 2024-01-29:
I don't know what's different between the r36s and r35s units.  I only have a r35s unit.

adotagarwal, 2024-01-29:
thanks a lot! 

r36s and r35s are identical i believe, some say r35s and some say r36s but when you order from aliexpress you can end up getting either. I would be happy to send you one though if it helps.

i am a developer and fairly handy with linux. would love to help add / offer support for this device. really appreciate you chiming in on this thread, and so quickly

christianhaitian, 2024-01-29:
Cool.  I'm not interested in adding support for the r36s.  Nothing wrong with the unit just that I don't want to support another clone of the rg351mp.  I am looking at the r33s since it's something I've been wanting for a few years now, a rk3326 unit in a bittboy size.  bittboy is what got me into this niche and eventually lead me to creating ArkOS.

christianhaitian, 2024-01-29:
I do support the r35s using the rg351mp image.

adotagarwal, 2024-01-29:
sorry to keep this going - but i noticed that my rice emulator has no video output. have you seen that before?

christianhaitian, 2024-01-29:
Yes if the resolution is not set correctly in the mupen64plus.cfg.  For the r35s/r36s it should be 480 which that backup should have set.

christianhaitian, 2024-01-29:
Try copying this video plugin within this zip file to `/opt/mupen64plus/` 

[mupen64plus-video-rice.zip](https://github.com/christianhaitian/arkos/files/14088575/mupen64plus-video-rice.zip)

christianhaitian, 2024-01-29:
[mupen64plus-video-rice.zip](https://github.com/christianhaitian/arkos/files/14088592/mupen64plus-video-rice.zip)

adotagarwal, 2024-01-29:
yeah that worked really well! thanks again. ive found rice works best for donkey kong 64 and the glidemk2 works best for episode 1 racer. 

im going to try a fresh image using both the rg351mp image here and the AeolusUX you linked earlier to see if either of those work right off the bat. i (subjectively) feel like the glidemk2 performance is a little worse than before, as there are stutters in star wars episode 1 racer