christianhaitian/arkos #931: Add videopac as a supported system

EnsignRutherford, 2024-01-30:
The Videopac is a clone of the Odyssey2.  Copying the odyssey2 es_systems.cfg section and tweaking for the videopac plays the same games.  Artwork is already available.

christianhaitian, 2024-01-30:
This doesn't interest me but if it's truly like supporting megadrive and genesis, it simple enough to add then.

EnsignRutherford, 2024-01-30:
Just for completeness, on my arcade build of used symbolic links to link the systems that were 'the same' because it just makes it more complete to me.  Like Turbografx = PCEngine, TurbografxCD = PCEngineCD, SegaCD = MegaCD.  The NES theme has all the artwork already.  By using symbolic links it doesn't take up any more/less space but allows people who recognize the systems to use the ones they want.  :)

EnsignRutherford, 2024-01-30:
```

   <system>
      <name>megacd</name>
      <fullname>Mega - CD</fullname>
      <path>/roms2/megacd/</path>
      <extension>.mdx .MDX .md .MD .smd .SMD .gen .GEN .cue .CUE .iso .ISO .sms .SMS .gg .GG .sg .SG .68k .68K .chd .CHD .zip .ZIP .7z .7Z .m3u .M3U</extension>
      <command>sudo perfmax %GOVERNOR%; nice -n -19 /usr/local/bin/%EMULATOR% -L /home/ark/.config/%EMULATOR%/cores/%CORE%_libretro.so %ROM%; sudo perfnorm</command>
      <emulators>
         <emulator name="retroarch">
            <cores>
               <core>genesis_plus_gx</core>
               <core>genesis_plus_gx_wide</core>
               <core>picodrive</core>
            </cores>
         </emulator>
         <emulator name="retroarch32">
            <cores>
               <core>picodrive</core>
            </cores>
         </emulator>
      </emulators>
      <platform>megacd</platform>
      <theme>megacd</theme>
   </system>
   <system>
      <name>videopac</name>
      <fullname>Philips Videopac G7000</fullname>
      <path>/roms2/videopac/</path>
      <extension>.bin .BIN</extension>
      <command>sudo perfmax %GOVERNOR%; nice -n -19 /usr/local/bin/retroarch -L /home/ark/.config/retroarch/cores/o2em_libretro.so %ROM%; sudo perfnorm</command>
      <platform>videopac</platform>
      <theme>videopac</theme>
   </system>
```

EnsignRutherford, 2024-02-01:
I also added AmigaCDTV which uses the same emulator as the AmigaCD32.  Artwork is already available.  Here's the es_systems.cfg entry:
```

   <system>
      <name>amigacdtv</name>
      <fullname>Commodore - Amiga CDTV</fullname>
      <path>/roms2/amigacdtv/</path>
      <extension>.chd .CHD .cue .CUE .ccd .CCD .lha .LHA .nrg .NRG .mds .MDS .iso .ISO</extension>
      <command>sudo perfmax %GOVERNOR%; nice -n -19 /usr/local/bin/amiga.sh %EMULATOR% %CORE% %ROM%; sudo perfnorm</command>
      <emulators>
         <emulator name="retroarch">
            <cores>
               <core>puae2021</core>
               <core>puae</core>
                          <core>uae4arm</core>
            </cores>
         </emulator>
         <emulator name="standalone"/>
         <emulator name="retroarch32">
            <cores>
               <core>uae4arm</core>
            </cores>
         </emulator>
      </emulators>
      <platform>amigacdtv</platform>
      <theme>amigacdtv</theme>
   </system>
```

closed 2024-02-24 by christianhaitian