christianhaitian/arkos #932: ALG games playing left justified on the RG503 screen

EnsignRutherford, 2024-01-30:
Hi, 

The singe.sh script has code I don't understand:

```
SINGEGAME=${basedir%.*}

function STDERR () {
        /bin/cat - 1>&2
}

if [ "$SINGEGAME" = "-fullscreen" ]; then
    FULLSCREEN="-fullscreen"
    shift
fi

if [ "$SINGEGAME" = "-blend" ]; then
    BLEND="-blend_sprites"
    shift
fi

if [ "$SINGEGAME" = "-nolinear" ]; then
    NEAREST="-nolinear_scale"
```

#SINGEGAME is the name of the game rom, e.g. "crimepatrol", "spacepatrol" so fullscreen, blend and nearest are never getting set as far as I can tell.  When -fullscreen is added to the singe command in the script it centers properly.

christianhaitian, 2024-01-31:
Interesting.  I'll address that in the next update.  Thanks for reporting the issue.

EnsignRutherford, 2024-02-15:
Here is the updated singe.sh that properly parses the command-line parameters.
[singe.zip](https://github.com/christianhaitian/arkos/files/14303045/singe.zip)

And here's the change to es_systems to pass the fullscreen parameters:

```
   <system>
      <name>alg</name>
      <fullname>American Laser Games</fullname>
      <path>/roms2/alg/</path>
      <extension>.alg .ALG</extension>
      <command>sudo perfmax %GOVERNOR%; nice -n -19 /usr/local/bin/singe.sh %ROM% -fullscreen; sudo perfnorm</command>
      <platform>alg</platform>
      <theme>alg</theme>
   </system>
```

closed 2024-02-24 by christianhaitian