christianhaitian/arkos #933: add usage retroarch menu on FN key on default configuration

VolodymyrDA, 2024-01-31:
In the latest update, they added the ability to use the FN button. According to me, the most logical thing to do is to assign the RetroArch menu button to it, allowing for a quick exit from games or changing settings. Here's how you can do it:

    In the RetroArch menu, go to Settings -> Input -> Hotkeys -> Menu Toggle (Controller Combo) and set it to L3 + R3.
    In RetroArch settings, navigate to Settings -> Input -> RetroPad Binds and assign the FN button (Button 16) to L3 and R3 since they are not used on our gamepad.
    To ensure RetroArch closes when you exit a game (Close Content), go to Settings -> User Interface -> Quit on Close Content and set it to ON.
    Save the configuration by going to RetroArch menu -> ArkOS -> Configuration File -> Save Current Configuration.

Please add modded configuration as default on r35\r36s оr others if you can.
Оr you add this solution to WIKI

christianhaitian, 2024-01-31:
Hello, this is not something I will add as default since these devices are a clone of the rg351mp which doesn’t have this button.  Perhaps this is something @AeolusUX would be interested in maintaining for the ArkOS build he maintains.

closed 2024-01-31 by christianhaitian

mrglennjones, 2024-01-31:
[heart]         Glenn Jones reacted to your message:

[Logo]<https://www.hwlinnovation.co.uk>
Glenn Jones
email: ***@***.***
web: www.hwlinnovation.co.uk<https://www.hwlinnovation.co.uk>
This message is intended only for the use of the person(s) ("the intended recipient(s)") to whom it is addressed. It may contain information which is privileged and confidential within the meaning of applicable law. If you are not the intended recipient, please contact the sender as soon as possible, and then delete it from your system; you should not copy the message or disclose its contents to anyone. HWL Innovation Ltd is a limited company registered in England and Wales. Company Registration No. 03758553 Registered Offices: 28 St Swithin Way, Andover, Hampshire, SP10 4NU

AeolusUX, 2024-01-31:
My image currently does this but instead of the FN button by itself It's Select+Start.
I've also been working on making it work for the standalone emulators and will add to my image soon. The only thing that I haven't done yet is Duckstation and PSP but if I find a way to change that I will add it soon in my image for the R35S/R36S.