christianhaitian/arkos #934: Multi disc games categorization

Domingster, 2024-01-31:
### Describe the Issue (If applicable)
Hello, this is maybe not bug, bud I can not figure out how to deal with multi disc Dreamcast games. I did categorize games thru folder where I had next folders (Disc 1, Disc 2, etc.) in older version, but now I did lost this option and I see under Dreamcast for example 2 or more same name files... Is possible to change it? Actually in Options I miss advanced and tools also but it is in options itself so that is not big deal.

### How can the issue be reproduced? (If applicable)
Start system :)

### What device are you using?
 RG351VS

### What version of ArkOS are you on?
27/1/2024

christianhaitian, 2024-01-31:
I'm not understanding your question.  Please clarify.

Domingster, 2024-01-31:
On my rom SD card i have 
\dreamcast\Game123\Disc1\Game123.bin......Game123.cue
\dreamcast\Game123\Disc2\Game123.bin......Game123.cue

When I open Dreamcast in ArkOS I see two:
Game123.cue
Game123.cue

But there is no folder structure.

christianhaitian, 2024-01-31:
While in the dreamcast system view, press select and then set Show Folders to `Having Multiple Games`

Domingster, 2024-01-31:
Sorry to annoy you. I was not able to find this option! Thank you!

closed 2024-01-31 by christianhaitian