christianhaitian/arkos #935: SSH/CLI access - ultimate goal of trying to get RTL8153 based USB Ethernet Adapter working

NeoMatrixJR, 2024-02-01:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

OK/Irrelevant

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?
Tried searching for this with no luck

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?
N/A
4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?
Yes
### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
Trying to gain access to CLI/SSH, ultimately, would like to know what it would take to get this to work with a USB Ethernet adapter based on the RTL8153 chipset.

### How can the issue be reproduced? (If applicable)

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

R35s

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
The last one from December.  I'm sorry, I don't have the device on me right now.

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2024-02-01:
Connect the usb ethernet device to the right otg port, connect the usb ethernet device to your network, then enable remote services. <https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rg351mp#q-how-do-i-ssh-into-arkos>

christianhaitian, 2024-02-01:
I use these 2 usb to ethernet adapters with my device without issue.

https://www.amazon.com/gp/product/B07G8VZ51Z/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
https://www.amazon.com/gp/product/B07SZTHPPR/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

NeoMatrixJR, 2024-02-01:
I tried to connect it.  It wouldn't show a network interface was added and the lights don't come on on the interface.  No link or activity.  Wanted to get dmesg logs from the device.

christianhaitian, 2024-02-01:
You can access cli directly on the device as well.
https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rg351mp#q-how-can-i-access-a-terminal-physically-on-ArkOS

christianhaitian, 2024-02-01:
Did you delete the ethernet connection while in the wifi configuration menu at all?

closed 2024-02-02 by NeoMatrixJR