christianhaitian/arkos #938: Switch to SD2 script doesn't make the system folders for roms like before.

AeolusUX, 2024-02-02:
### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
The switch to sd2 script still doesn't work right after the last update. I just got a new sd card and tried to use it but when I used switch to sd2 for roms the process is fairly quick. When I inserted the sd card into my pc it didn't do all the folders for roms, I've seen I think just 10 of them there so I immediately reformat the card and did the process again using a different format card and it still didn't make the system folders like it does before. 

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)
R35S

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
1272024-1

### Anything else you'd like to include that may help to review this issue or feature request?

AeolusUX, 2024-02-02:
And it won't switch to main sd.

christianhaitian, 2024-02-02:
Sounds like the `roms.tar` file is missing from the root `/` folder.  Did you delete this file from your install?  What error are you getting when switching to main sd?

christianhaitian, 2024-02-02:
switch to main sd works for me and all pertinent folders are created as needed.  I can't replicate this issue.

AeolusUX, 2024-02-02:
Alright I'll just do a fresh from 351mp image and see how it works. Might've chosen the wrong sd card with updated arkos.

AeolusUX, 2024-02-02:
Yeah seems like the roms.tar was deleted my bad I have no idea how that happened xD

AeolusUX, 2024-02-03:
Seems like firstboot.sh was deleting the roms.tar file.

AeolusUX, 2024-02-03:
Found that the expand file was deleting the roms.tar file. 
Checked the expand script and it looks for rk3326-rg351mp-linux.dtb.
Added it back and now it doesn't delete it anymore.
Posting here for reference when someone deletes their rk3326-rg351mp-linux.dtb.

closed 2024-02-03 by AeolusUX