christianhaitian/arkos #942: Internal scraper not working for Gamecube on RG353V

dani7959, 2024-02-03:
Selecting Gamecube system for scraping results in "Warning: some of your selected systems do not have a platform set. Results may be even more inaccurate than usual!".

Using scrape for individual Gamecube games doesn't seem to perform any search.

Checked game names on screenscraper.fr
Checked /etc/emulationstation/es_systems.cfg

christianhaitian, 2024-02-03:
This unit is not good for GC except for maybe less than a handful of games.  So I can’t imagine that someone will realistically play many GC games on this.  Nonetheless, I’ll review this when possible.

christianhaitian, 2024-02-04:
The platform name in `/etc/emulationstation/es_systems.cfg` need to be changed from `gamecube` to `gc` to resolve the issue.  This will be resolved in the next update.

closed 2024-02-24 by christianhaitian