christianhaitian/arkos #943: PPSSPP menu DPI

dani7959, 2024-02-03:
The PPSSPP menu looks OK except for Control mapping which is wider than the display.  The remove mapping button is not visible.

Settings > Controls > Control mapping

I think the DPI should be adjusted.

christianhaitian, 2024-02-03:
This is due to the screen being 4:3.  Not something I can fix as it’s a limitation of the emulator as it’s not designed for use with this aspect ratio.

dani7959, 2024-02-03:
Thanks, I'll try to report the issue upstream.

Can you please tell me what PPSSPP version are you using? Settings > Tools > System information > Device info > PPSSPP build just reads "Release".

Also, is the PPSSPP updater working? There is an option at Settings > System > Check for new versions of PPSSPP.

christianhaitian, 2024-02-03:
Should be 1.16.6.  The changelog says specifically. That updater will not work here.