christianhaitian/arkos #944: Emulationstation doesn't save gamelist.xml on R3+power

Sonic803, 2024-02-03:
When shutting down with R3+power some times the favorites and last played games don't get saved by emulationstation.

Possible fix:
modify pause.sh,finish.sh to kill emulationstation before poweroff

example finish.sh

```
#!/bin/bash
if [ ! -e "/home/ark/.config/.SWAPPOWERANDSUSPEND" ]; then
  pkill emulationstatio && while pgrep -l emulationstatio; do sleep 1;done;
  sudo systemctl poweroff
else
  sudo systemctl suspend
fi
```

R36s
2.0 01272024-1

christianhaitian, 2024-02-03:
If you're shutting down while in game and not after you return to the ES menu, this will happen.  The shutdown process stops the ED systemd process so your suggested fix won't work or just cause other issues.  To properly update the ES menu, do the quick shutdown while you're in the ES menu.

christianhaitian, 2024-02-03:
Also note that I don't support the r36s units but I can answer basic questions like this.

closed 2024-02-03 by christianhaitian