christianhaitian/arkos #945: Add artwork for full-size screen gameplay for Tiger LCD games

EnsignRutherford, 2024-02-03:
I found a set of artwork files that were made for 44 Tiger LCD games.  The show the whole handheld on the screen and resize it so that it's too small to play.  I updated the layout files so now on the RG353V and RG503 they run full screen and just show the background so game play is much better than just the games themselves.  The file is too large to upload here.  Is this something that should be included in ArkOS?

christianhaitian, 2024-02-03:
I'm aware of them but won't include them as it would increase the size of the image which I don't want to do, however, if they're available from a website somewhere, i'll be happy to link to them from the wiki entry for TigerLCD.

EnsignRutherford, 2024-02-04:
Let me see if I can find a place to share.  I put together the overlays, the proper gamelist.xml and images of each game.  Supports 44 games in all.

Quick question: I spent a little time trying to get TMNT Shredder's Revenge working on the latest build of ArkOS.  Mono crashses after it's generating the gamedata.  I know this is prolly an old question but has anyone got it to work?

christianhaitian, 2024-02-05:
Yes there was a fix for that.  Check the Portmaster Discord>Testing-n-dev>TMNT Shredder's Revenge thread.  

Alternatively, you can try this workaround. https://github.com/christianhaitian/arkos/issues/777#issuecomment-1693758677

christianhaitian, 2024-02-05:
btw, I can probably store the overlays on this git as well.

EnsignRutherford, 2024-02-05:
It's actually a lot bigger than I first mentioned... 528MBs for 44 backgrounds.

https://drive.google.com/file/d/1bJaBW6J3nPW_-PDEq_mXwe49kVqopKhE/view?usp=drive_link

christianhaitian, 2024-03-02:
I've finally gotten around to download and make these backgrounds available on the git.  I've also credited you for sourcing these files and credited the creators here: https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information#tiger-lcd-games-available-only-on-the-rk3566-devices-such-as-the-rg353m-only

Thanks again for the find.

closed 2024-03-02 by christianhaitian

jborza, 2024-04-04:
Hi @christianhaitian ,
I'm a but flummoxed, but trying to find this "Testing-n-dev" [thread](https://github.com/christianhaitian/arkos/issues/945#issuecomment-1926010510) , but can't find it now - is it because of nintendo and lawsuits?

christianhaitian, 2024-04-04:
> Hi @christianhaitian , I'm a but flummoxed, but trying to find this "Testing-n-dev" [thread](https://github.com/christianhaitian/arkos/issues/945#issuecomment-1926010510) , but can't find it now - is it because of nintendo and lawsuits?

You should ask in the PortMaster Discord what happened to it as I'm not sure.