christianhaitian/arkos #947: Now that TRS-80 is supported via LibRetro Mess is it possible to get support added for other systems supported by Mess such as the BBC Micro?

SuperBadger77, 2024-02-05:
Hi,

I have previously asked for BBC Micro support to be added with a previous request and know that it was investigated but it never went anywhere. Now that TRS-80 support has been added, does this open the door to other systems to be supported such as the BBC Micro or Acorn Electron? I know it is supported in distributions such as Batocera (MAME and LR-MESS), and RecalBox:
https://wiki.batocera.org/systems:bbc
https://wiki.recalbox.com/en/emulators/computers/bbc-micro/beebem
If any testing or help required with setup and I can help out with that.

christianhaitian, 2024-02-05:
I'll eventually take a look at this as I have some interest in BBC Micro.  I will say that TRS-80 works far better on the standalone XRoar emulator than through MESS due to control issues.  Hopefully performance and control with MESS is ok with BBC Micro.

SuperBadger77, 2024-02-06:
Well when you do get round to it let me know if you need any help / support / testing.
Here's a great website for BBC Micro games: https://bbcmicro.co.uk//index.php You can download for free or play in browser (legally), plus there are loads of home brew games, and nearly every game someone has edited them to add info at the start containing the controls and any back story / information before playing :-)

christianhaitian, 2024-02-06:
Awesome.  Thank you

EnsignRutherford, 2024-02-13:
I had some time this morning so I was able to get mess running the bbcb emulation, load a rom, but no idea how to get it to autostart the rom.  I was using the TRS-80 emulation as a starting point.  BIOS files go into the /roms/bbcb folder.

EnsignRutherford, 2024-02-13:
BIOS Files go in /roms/bbcb:
bbc_acorn8271.zip
bbcb.zip
saa5050.zip
HASH files go in /roms/bios/mame/hash:
bbcmc_flop.xml
bbc_cass.xml
bbc_flop_6502.xml
bbc_flop_32016.xml
bbc_flop_68000.xml
bbc_flop_80186.xml
bbc_flop_arm.xml
bbc_flop_hybrid.xml
bbc_flop_torch.xml
bbc_flop_z80.xml
bbc_hdd.xml
bbc_rom.xml
bbcb_flop.xml
bbcb_flop_orig.xml
bbcb_flop_us.xml
bbcbc.xml
bbcm_cart.xml
bbcm_flop.xml

lookup a game you want to run in those files and save it with the right name for the zip and the rom.

The command to run:
/usr/local/bin/retroarch -v -L /home/ark/.config/retroarch/cores/mess_libretro.so /roms2/bbcb/chuckie.zip

It starts but ends up at the BASIC prompt.  No idea what to do to get it to run.

SuperBadger77, 2024-02-13:
> BIOS Files go in /roms/bbcb: bbc_acorn8271.zip bbcb.zip saa5050.zip HASH files go in /roms/bios/mame/hash: bbcmc_flop.xml bbc_cass.xml bbc_flop_6502.xml bbc_flop_32016.xml bbc_flop_68000.xml [...]

Great - I'll try and recreate and see if I can get any further. I've added a BBC entry to ES before so hopefully that should be straight forward. If you are using an .ssd file (single sided disk) then in Batocera it automatically runs the following 2 commands:
*CAT (to list the disk contents)
then after 3-4 carriage returns it does:
*EXEC !BOOT
Most ssd files should have a !BOOT file on them. If it's tape then you can use the CHAIN"" command or put the program name in the "".
Anyways I'll see if I can get any further.
Most of my games have pad2key files to map the controls to the joypad so will need to check if ArkOS has anything similar or I'll connect a keyboard.

EnsignRutherford, 2024-02-13:
I don't know if mess allows to send a command-line into the basic processor.  xroar for the Coco allows this.  I don't have access to Batocera so if you do and can send me the script that starts a bbx micro game I can tailor it to our needs.

christianhaitian, 2024-02-13:
oh good point.  I forgot about that.  Might need to use the alternative mapping workaround we used for MESS.

EnsignRutherford, 2024-02-13:
Using the input remapping would work for controlling as well as gptokeyb, but the problem is STARTING the rom.  At least mess autoruns the 'cartridge rom' for the coco.  I tried a bbc rom file and it stays at the basic prompt.

EnsignRutherford, 2024-02-13:
This could also be promising:

https://haprpi.blogspot.com/p/the-bbc-lives.html?m=1

SuperBadger77, 2024-02-13:
> This could also be promising:
> 
> https://haprpi.blogspot.com/p/the-bbc-lives.html?m=1

I've seen that page in my years trying to get the BBC emulator to work. The Rick Dangerous build has the BBC emulator working and you can map keys to pad in MAME. I tried compiling various BBC emulators on ArkOS a while back but didn't ever get it to fully compile - but my skillz are severely lacking in the unix \ compiling area - if you can get it working then that would be great.

SuperBadger77, 2024-02-13:
> gptokeyb is available for mapping controls. You can check `/usr/local/bin/coco.sh` for an example of how it's used

If it works like this then converting my files should be fairly easy and I can test it (depending on if the command line issue mentioned above is an issue).
[StarQuake (1987) (Bubble Bus Software).ssd.p2k.zip](https://github.com/christianhaitian/arkos/files/14271606/StarQuake.1987.Bubble.Bus.Software.ssd.p2k.zip)
Here's an example. I can just do a global search and replace and also remove the comments.
0:left = Z ;; Left
0:right = X ;; Right
0:up = s11 ;; Up '*' Pick Up Object
0:down = s07 ;; Down '?' Lay Bridging Platform
should become:
left = Z
right = X
up = *
down = /

SuperBadger77, 2024-02-13:
> I don't know if mess allows to send a command-line into the basic processor. xroar for the Coco allows this. I don't have access to Batocera so if you do and can send me the script that starts a bbx micro game I can tailor it to our needs.

Here's the issue that added in BBC support: https://github.com/batocera-linux/batocera.linux/pull/5998

SuperBadger77, 2024-02-13:
Actually might have been this one: https://github.com/batocera-linux/batocera.linux/pull/4926

SuperBadger77, 2024-02-13:
Here is the code where it loads a game: https://github.com/batocera-linux/batocera.linux/pull/4926/commits/68ee022c3618125e3032dd7d337778fd57274de5
                # bbc has different boots for floppy & cassette, no special boot for carts
                if system.name == "bbc":
                    if system.isOptSet("altromtype"):
                        if system.config["altromtype"] == "cass":
                            commandArray += [ '-autoboot_delay', '2', '-autoboot_command', '*tape\\nchain""\\n' ]
                        elif left(system.config["altromtype"], 4) == "flop":
                            commandArray += [ '-autoboot_delay',  '3',  '-autoboot_command', '*cat\\n*exec !boot\\n' ]
                    else:
                        commandArray += [ '-autoboot_delay',  '3',  '-autoboot_command', '*cat\\n*exec !boot\\n' ]

SuperBadger77, 2024-02-14:
Just added an entry to es_systems.cfg:
	<system>
		<name>bbcm</name>
		<fullname>BBC Micro</fullname>
		<path>/roms2/bbcmicro/</path>
		<extension>.d77 .d88 .dsk .img .ssd .bbc .dsd .adf .ads .fsd .wav .tap .bin .zip .7z .D77 .D88 .DSK .IMG .SSD .BBC .DSD .ADF .ADS .FSD .WAV .TAP .BIN .ZIP .7Z </extension>
		<command>sudo perfmax %GOVERNOR%; nice -n -19 /usr/local/bin/retroarch -v -L /home/ark/.config/retroarch/cores/mess_libretro.so %ROM%; sudo perfnorm</command>
		<platform>advision</platform>
		<theme>advision</theme>
	</system>

And added a rom and bios but when I try and run anything it goes back to the main menu - where are the logs written to? @christianhaitian

SuperBadger77, 2024-02-14:
Ignore that - used the command line and seeing the errors now.

SuperBadger77, 2024-02-14:
@EnsignRutherford can you share the uef chuckie.zip file you used? For some reason the list in the hash files has very little in the way of floppy games. Which is weird as Batocera has the same bios files, and most games are on floppy / .ssd. I need .uef games but don't have any. At the moment I can't even get to the BBC prompt.

SuperBadger77, 2024-02-14:
OK finally got somewhere. Found all the tape images here: https://www.stairwaytohell.com/bbc/index.html?page=tapeimages
Finally got to the command prompt with chuckieg.zp attached
[chuckieg.zip](https://github.com/christianhaitian/arkos/files/14288461/chuckieg.zip)
However, unable to type CHAIN "" and press return as my USB keyboard seems to have all the hotkeys for retroarch. As soon as I type 'H' it resets the machine.

SuperBadger77, 2024-02-14:
OK - sorry about the live blog but H was assigned to reset in RetroArch so removed that. If I run:
/usr/local/bin/retroarch -v -L /home/ark/.config/retroarch/cores/mess_libretro.so /roms2/bbcb/chuckieg.7z
I get the BBC prompt - if I type in CHAIN "" it says BAD PROGRAM. So not sure why that is failing.

EnsignRutherford, 2024-02-14:
Sounds like it didn't load it as a cassette.

SuperBadger77, 2024-02-14:
Holy smokes, a bit of googling and it seems you need to set it to point to the tape drive if it has multi tape / disc systems.
*TAPE
CHAIN ""
She's alive and booting!!!!

EnsignRutherford, 2024-02-14:
So we're back to how to send those commands into a cassette program versus a disk versus a rom

SuperBadger77, 2024-02-14:
Yeah, not sure how we do all that. It loaded the tape, I removed the reset hotkey, then I had to press K to reassign the keys to ones that didn't do anything in RetroArch. Once I'd done that it seemed to play pretty well with the USB keyboard. No notable slow down. Probably have to created key to pad mappings for each game. Not looked at that bit yet. But here's a couple of photos of it running and me getting to level 2 :-)
[image]
[image]

EnsignRutherford, 2024-02-14:
Wow looks good... so close.

EnsignRutherford, 2024-02-14:
I'm now thinking the changes need to be made to mess itself.  mess autostarts Coco roms when they are loaded.

EnsignRutherford, 2024-02-17:
I've researched this as much as I can and I really think without modifying the bbcmicro core to auto run programs it's not going to work within a console build.

christianhaitian, 2024-02-17:
Too bad.  Perhaps someone will port one of the other BBC Emulators to one that could be used on here.

EnsignRutherford, 2024-02-18:
I've been reviewing the MESS core with a hex editor and it supports the "autoboot_delay" and "autoboot_command" functions.  I'm just not sure how to use them on a properly formatted command-line.

SuperBadger77, 2024-02-18:
Just googled and found this:

https://forums.libretro.com/t/guide-play-non-arcade-systems-with-mame-or-mess/17728/41

EnsignRutherford, 2024-02-18:
I need a resource that can state what's the right format for this command:

/usr/local/bin/retroarch -v -L /home/ark/.config/retroarch/cores/mess_libretro.so /roms2/bbcb/chuckieg.zip -autoboot_delay "2" -autoboot_command "*TAPE\n\nCHAIN ""\n"

I've tried various combinations but to no avail.  The "autoboot_delay" and "autoboot_command" are processed within mess_librretro.so, NOT retroarch so the command-line is not formatted properly.

Oberworld, 2024-02-23:
I don't know if this helps, but I have some bbc micro ssd games working on windows under mame.
What I have are bat files for each game, then from es_systems.cfg I run the bat file.
The content of the bat file is this:
`mame64.exe bbcbp -flop1 ".\roms\bbcb_flop\3DBombAlley_BBC.ssd" -autoboot_delay 2 -autoboot_command "CHAIN"""LOAD""\n`

Probably using lr-mame (Current) the command line is very similar.
I've found this info:
https://forums.launchbox-app.com/topic/54987-autoboot-command-script-for-mame-swl-computer-systems/

>  You can accompish this either by using a command line or by editing the electron.ini in...system\mame\ini\ (if you use retroarch:
> 
> Example:

```
#
# SCRIPTING OPTIONS

autoboot_command          "*cat\n\n\n\n\n\n*run !boot\n"
autoboot_delay            2
autoboot_script  
```      
This example is for electron but maybe you can modify for bbc micro and create a bbcb.ini.

I hope this helps, regards.

christianhaitian, 2024-04-01:
What's the current status of this?  Anything useful worth reviewing yet?

EnsignRutherford, 2024-04-03:
Actually I gave up on the BBC Micro as there was no easy way to get keyboard input into it without rewriting the emulator itself to support it.

 

If you have any ideas I’m all ears!

 

ER

christianhaitian, 2024-04-03:
Ok.  I'll close this issue for now and hopefully a solution comes up in the near future that doesn't need a keyboard attached like what was found for coco3 and apple ii emulation.

closed 2024-04-03 by christianhaitian