christianhaitian/arkos #948: Manually adding additional emulation section for rom hacks (feature request)

spamsool, 2024-02-05:
Hello

I would kindly like to ask about the status of the rom hacks support in ArkOS.
I have checked the closed issues and I am aware that there is a list of additional emulation sections to be added in the future.
It is mentioned in the issue 193 https://github.com/christianhaitian/arkos/issues/193 and marked as completed.

However in the meantime would it be possible that user could somehow add it itself? I play a lot of rom hacks and I would be mostly interested in adding GB hacks, GBC hacks, GBA hacks, NES hacks and SNES hack (SNES hacks is not on the list in the issue 193).

I could use JelOS for this which has rom hacks support already implemented (I checked on Anbernic RG353V running JelOS) but I also use R32S which is a great little console running only ArkOS. I simply prefer using ArkOS to any other console linux distros.

Maybe it could be possible to somehow reuse the solution of rom hacks sections from JelOS to make it a bit easier?

Sorry to bother you about this feature request, but I just love rom hacks! :-)
Thanks for your help :)

christianhaitian, 2024-02-06:
Why not just add rom hacks to a subfolder within the existing systems?

spamsool, 2024-02-06:
Thank you christianhaitian for your quick reply.

I know it is very subjective, but keeping right order in rom collections is a important topic for many people including myself.
Rom hacks usually have the same or very similar name name to original titles, so they could me easly mistaken when chosen.
The amount of the rom hacks files is minimal compared to number of roms in the original system directory - often they remaing difficult to locate and to distinguish from originals. - moving them to a separate category in the menu just makes everything cleaner tidier and easier to use or manage.
It generally is just a matter of aesthetics and personal preferences. Rom hacks kept in separate category in the menu would just simplify things. There are also separate folders on SD card structure for rom hacks ie. nesh snesh gbh gbch and gbah (and similar for Sega systems) - and Emulation station could create separate catergory from this folders like it is now in JelOS.

That is why we have different systems kept in separate folders in first place. Adding rom hacks to separate menu would add this finishing touch to ArkOS game management support.
Others would rather put all roms in one common folder and never look back.

Thank you for your understanding.

Greg

christianhaitian, 2024-02-06:
Ok.  It is mostly a personal preference as organization can still be achieve with sub folders.  Not something I will universally add at this time, however, you can manually add such entries in `/etc/emulationstation/es_systems.cfg`. Make sure to create a backup copy of this file before you edit it.  Also keep a backup copy of your edited version in case a future update impacts it which should be a rare occurrence if at all.

closed 2024-02-06 by christianhaitian