christianhaitian/arkos #949: Show stderr for Options scripts

SirNate0, 2024-02-06:
Currently `/etc/es_systems.cfg` uses 
```xml
<system>
	<name>options</name>
	<fullname>Options</fullname>
	<path>/opt/system/</path>
	<extension>.sh .SH</extension>
	<command>sudo chmod 666 /dev/tty1; %ROM% > /dev/tty1; printf "\033c" >> /dev/tty1</command>
	<platform>ignore</platform>
	<theme>retropie</theme>
</system>
```

This redirects only stdout from the script to the tty. Could we show stderr as well (`2>&1`), or do some of the existing scripts expect *only* stdout to be shown to the user?

christianhaitian, 2024-02-06:
I see no harm in that.  What you working on that you're asking for this?

SirNate0, 2024-02-06:
Generally, I'm trying to use python raylib bindings to create some tools (specifically, I'm trying to create a map editor for the games I make as a hobby) . So not something that will generally benefit others directly, though should someone else try making their own option scripts the stderr redirection would probably be helpful, especially in debugging them. (It was obvious for me that there was a missing error message - no graphics showed up and the program exited on its own in the middle of the printed text - it may not be obvious for other scripts that are, for example, all text based)

EnsignRutherford, 2024-02-15:
```
<system>
	<name>options</name>
	<fullname>Options</fullname>
	<path>/opt/system/</path>
	<extension>.sh .SH</extension>
	<command>sudo chmod 666 /dev/tty1; %ROM% > /dev/tty1; printf "\033c" 2&gt;&amp;1 &gt;&gt; /dev/tty1</command>
	<platform>ignore</platform>
	<theme>retropie</theme>
</system>
```