christianhaitian/arkos #950: Root Filesystem Full

EnsignRutherford, 2024-02-07:
Latest ArkOS on RG503:

Trying to install RVGL from PortMaster started to get write failed errors.  Checking the filesystem free space:

Filesystem      Size  Used Avail Use% Mounted on
udev            469M     0  469M   0% /dev
tmpfs            97M  2.3M   95M   3% /run
/dev/mmcblk1p4  9.9G  9.8G     0 100% /
tmpfs           482M     0  482M   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           482M     0  482M   0% /sys/fs/cgroup
/dev/mmcblk1p3   99M   32M   68M  32% /boot
/dev/mmcblk1p5   19G  267M   19G   2% /roms
/dev/mmcblk2p1  932G  848G   85G  91% /roms2
tmpfs            97M     0   97M   0% /run/user/1002

root is full?!?!?

Any idea what might be taking up so much space?

christianhaitian, 2024-02-07:
Have you been doing apt installs or doing other builds?

EnsignRutherford, 2024-02-07:
Just installing some ports.  Haven't installed anything via apt or building anything other than TMNT.  I just noticed now the zero space available.  I'm using du/df to try and track down where the space is being used.

christianhaitian, 2024-02-07:
Check for for a swapfile in the root partition.  maybe rvgl did that for some reason.

EnsignRutherford, 2024-02-07:
Found it.  /root/.rvgl = 4G!!!!!!

EnsignRutherford, 2024-02-07:
sudo du -xm / | sort -rn > usage.txt

This will show every folder and the size it has allocated to it sorted.  That's how I found it.

EnsignRutherford, 2024-02-07:
Sorry for the red flag.

closed 2024-02-07 by christianhaitian

EnsignRutherford, 2024-02-07:
Rats! While I have your attention.  :) Thanks for letting me contribute to this.  Looks like I'm going to be buying the next Ambernic that is supposed to play PS2 when it's released in March.

QQ: I REALLY want to add a MUTE button/key sequence to the RG503.  Can you point me in the right direction?

christianhaitian, 2024-02-07:
<https://github.com/christianhaitian/ogage/blob/rg503/src/main.rs>
Modify the source I link above to do this.
You probably want something like this line: <https://github.com/christianhaitian/ogage/blob/30e6c11bd35c66494272f742e6d9f178e47cc2e7/src/main.rs#L83>

contents of mute_toggle.sh
```
if [[ -z $(pgrep kodi) ]]; then
  cur_vol="$(/usr/local/bin/current_volume)"
  if [[ "$(tr -d '\0' < /proc/device-tree/compatible)" == *"rk3326"* ]]; then
    DEVICE="Playback"
  else
    DEVICE="Master"
  fi

  if [ -f /dev/shm/VOLUME_BEFORE_MUTED ] && [ "$cur_vol" = "0%" ]; then
    amixer -q set ${DEVICE} $(cat /dev/shm/VOLUME_BEFORE_MUTED)
    rm -f /dev/shm/VOLUME_BEFORE_MUTED
  else
    echo $cur_vol > /dev/shm/VOLUME_BEFORE_MUTED
    amixer -q set ${DEVICE} 0
  fi
fi
```

EnsignRutherford, 2024-02-07:
I followed the instructions at https://github.com/christianhaitian/ogage/tree/rg503 in order to compile ogage but it fails compiling with these errors:

```
error[E0658]: `cfg(target_has_atomic)` is experimental and subject to change
    --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/log-0.4.20/src/lib.rs:1356:7
     |
1356 | #[cfg(target_has_atomic = "ptr")]
     |       ^^^^^^^^^^^^^^^^^^^^^^^^^
     |
     = note: for more information, see https://github.com/rust-lang/rust/issues/32976

error: aborting due to 9 previous errors

For more information about this error, try `rustc --explain E0658`.
error: could not compile `log`.
```

christianhaitian, 2024-02-07:
What environment are you using to build this?

EnsignRutherford, 2024-02-07:
On the RG503, under ArkOS.

christianhaitian, 2024-02-08:
The version of rust available through the Ubuntu repo there is old.  Do the following:

`sudo apt remove -y rustc`
`curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh` (Do the customize install, choose the default options, but for profile do minimal.)
`source "$HOME/.cargo/env"`

Then try to build ogage again.

EnsignRutherford, 2024-02-08:
How on Earth did you figure that out?  It worked!

christianhaitian, 2024-02-08:
I've dealt with compiling rust enough to know when it's out of date, it will throw weird compiling errors since it downloads assets with every new build.

christianhaitian, 2024-02-08:
KEY_PLAYPAUSE is only available from a USB C headphone dongle.  I added that for use with USB C headphone dongles.  Possible key values for devices can be found via `evtest` for the `retrogame_joypad` controller on your rg503.

```
Input device name: "retrogame_joypad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 312 (BTN_TL2)
    Event code 313 (BTN_TR2)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
    Event code 544 (BTN_DPAD_UP)
    Event code 545 (BTN_DPAD_DOWN)
    Event code 546 (BTN_DPAD_LEFT)
    Event code 547 (BTN_DPAD_RIGHT)
    ```

EnsignRutherford, 2024-02-08:
Gotcha.  So you're using BTN_THUMBR as the global hotkey.  So hitting R3 and L3 could be used to mute.

EnsignRutherford, 2024-02-08:
Sorry to ask you a question about your code, but I'm at a real loss (not running out of space loss).  The main.rs defines the hotkey as BTN_TRIGGER_HAPPY5, but we know BTN_THUMBR is the global hotkey.  Why does that work?  I haven't started testing yet.

christianhaitian, 2024-02-09:
Sounds like you're in the wrong branch.  The rg503 branch has no BTN_TRIGGER_HAPPY5 defined in the main.rs source file.

EnsignRutherford, 2024-02-09:
Cr*p... never mind.... yeah wrong branch.  I got it now!

EnsignRutherford, 2024-02-09:
Thanks for your help... got it working perfectly.  :)  The changes of course were very simple.