christianhaitian/arkos #952: dualshock controller not working in PCSX_REARMED_RUMBLE on RG353VS

dani7959, 2024-02-08:
Trying to set the Port 1 device type to dualshock controller and enjoy rumble :)

Using PCSX_REARMED_RUMBLE (Retroarch32):
- thumbsticks not working
- rumble not working as per the [wiki](https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rg353v#q-how-do-i-enable-rumblevibration-in-pscx_rearmed)

Also tested DUCKSTATION (Retroarch) and thumbsticks are working on dualshock (rumble not working and probably not supported)

christianhaitian, 2024-02-08:
Try updating the PCSX_REARMED_RUMBLE core via retroarch32 and try it again.  Duckstation does not support rumble currently.

dani7959, 2024-02-08:
Core seems to be up-to-date, current version r21.
Re-downloaded databases, controller profiles etc from the update menu.
Using bios psxonpsp660.bin.
Tested with and without the optional bios files (scph5500.bin scph5501.bin scph5502.bin) which are correctly detected by the PCSX_REARMED_RUMBLE core.

In all scenarios the dualshock controller profile is not working correctly (i.e. analog input not working, rumble not working).

The hardware is tested working on other platforms (both analog controls and rumble).

christianhaitian, 2024-02-09:
I tested this core using Mr. Driller and rumble worked fine.  I did notice I had to press L1+R1+Select to enable analog mode.  Must have been a recent addition to the core.

dani7959, 2024-02-09:
Indeed. The analog toggle key combo can even be remapped. The option is right below the rumble toggle, under Core Options > Input. I don't know how I missed it.
As for rumble, I tested some more roms and it does work.

Thanks a lot - I will close the issue.