christianhaitian/arkos #955: OpenGL error messages on ArkOS on RG503

EnsignRutherford, 2024-02-11:
Hi,

While testing alephone with a custom game, e.g. "Marathon Eternal" running it from SSH:

./alephone "/roms2/ports/alephone/Marathon Eternal"

it generates this error:

[ogg @ 0x5592d99040] 1922 bytes of comment header remain
OpenGL (Shader) renderer is not available (screen.cpp:971)
WARNING: Failed to initialize OpenGL renderer
WARNING: Retrying with Software renderer

It then runs using the Software rendered.

Looks like a shader issue with OpenGL, from what I've been able to read about it.

When I Ctrl+C out of alephone I get 8 instances of this message:

arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '3'.

It also happens when I exit PortMaster.  Appears onscreen before the screen goes back to ES.

Not sure what's causing this.

ES

christianhaitian, 2024-02-11:
There's no OpenGL available.  Just OpenGLES.  The messages are just harmless libmali messages

closed 2024-02-11 by christianhaitian

EnsignRutherford, 2024-02-11:
I thought so.  Hate the messages though.  :) Thanks!