christianhaitian/arkos #956: No Theme, when transfering files/ Game ROMSs via FTP

OberfragGER, 2024-02-12:
I recently brought a R36S with ArkOS.

Now i updated the device with the last release (01/27/2024-1). After pushing the latest Portmaster to the device via FTP... the UI-Theme is not working any longer. I deleted the file (install Portmaster.sh) and everything runs fine again.

No i uploaded some roms via ftp and now it stopped working again.

Unter the Settings (Start button i have in the UI-Section NO Theme option- the first 2 lines are just missing). Also the standard-theme is not working. It looks very strange and the letters are not highlitetd. So its ver hard to switch inside the ROMs folder e.g.

Do you have any hints how to enable again the working themes?

christianhaitian, 2024-02-12:
All theme files are located on SD1 in the `/roms/themes` folder.  It sounds like to me your theme files are missing or corrupted.  A few tips, 

1.  On any of these devices, do not keep the stock OS card that comes with them.  They tend to fail quickly.  Get a name brand sd card such as Samsung or Sandisk for the OS card for reliability.
2. I don't officially support the R36S handheld.  There is a community maintained version of ArkOS available for the R35s/R36s.  You can find more information about it here: https://github.com/AeolusUX/ArkOS-R3XS
3. There are other CFW options available for the R35s/R36s as well such as AmberElec and JELOS.  You can find more information about them by searching via Google or joining the Retrohandheld Discord via the invitation link below:
https://discord.gg/retrohandhelds

Hopefully these help.  Good Luck.

closed 2024-02-12 by christianhaitian

OberfragGER, 2024-02-13:
Thanks a lot. That was the case. Somehow... the themes gets deltetd when im pushing some files to the card.
I will reinstall ArkOs.

Your hint helped me a lot!