christianhaitian/arkos #959: Wolfenstein games doesn't recognize SELECT+START to exit

EnsignRutherford, 2024-02-14:
Latest version of ArkOS, ecwolf.sh doesn't call gptokeyb in order to allow exit.  I have  aversion that has the necessary changes.  Should it be added to the build?

christianhaitian, 2024-02-14:
It's something I've been meaning to add.  I just haven't yet.

EnsignRutherford, 2024-02-14:
This has the necessary changes.
[ecwolf.zip](https://github.com/christianhaitian/arkos/files/14285166/ecwolf.zip)

christianhaitian, 2024-02-14:
You better test and make sure you can reassign buttons in the emulator.  Sometimes gptokeyb interferes with that for emulators.  That's why I tend to go with a python hotkey daemon instead for things like this.

EnsignRutherford, 2024-02-19:
Wow good catch.  Time for me to write a python script.

christianhaitian, 2024-02-20:
I'm going to use a python 3 script to do it instead like I've done for other emulators.  Eventually, I'd like to use just one global python 3 script for all of emulators that need a exit keydemon and just pass the name of the executable that needs to be killed as a variable.  This script would have to account for devices that use a different combination for killing programs such as the rgb10 which does it's plus key + start and the Chi which uses the 1 key + start.  That's a project for the future though.

EnsignRutherford, 2024-02-20:
If you don't mind I can take a crack at the generic python handler for this?

closed 2024-02-24 by christianhaitian