christianhaitian/arkos #968: Add script that facilitates using both SD Cards for roms storage on r36s

VolodymyrDA, 2024-02-24:
Please add this script from AeolusUX github for r36s

https://github.com/AeolusUX/ArkOS-R3XS/releases/tag/ArkOS.V2.0.02062024
"Added script that facilitates using both SD Cards for roms storage.
Go to Options -> Advanced -> Switch to SD2 for Roms, Run the script and wait for it to finish.
There should be a new item available at Options -> Advanced -> Read from SD1 and SD2 for Roms run it and it will restart Emulation Station and now show all roms from both cards.
(Thanks to: christianhaitian for pointing out how to do it and kc234567890 for making the initial config that makes this possible for all systems.)
"

christianhaitian, 2024-02-24:
Not something I will officially support. You’re welcome to add this to your own setup though.

closed 2024-02-24 by christianhaitian

VolodymyrDA, 2024-02-24:
Why you not added this usefull feature?
I  do not know how add this sctipt by myself :(

christianhaitian, 2024-02-24:
I'm not interested in it.  Contact AeolusUX for assistance on how to replicate this on your end.