christianhaitian/arkos #969: Powkiddy RGB30 display not working / HDMI output is fine

mx-bernhard, 2024-02-25:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANce FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not the Stock OS that comes with your device, AmberElec, JelOS, or UnOfficialOS.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and these other distributions is different with different capabilities.  If there are features you prefer with the other available distributions, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions section available for your supported device in the wiki?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)

ArkOS boots on my RGB30 when using the HDMI connector and my old full hd tv and I see the EmulationStation frontend. Without HDMI it also seems to boot, but the screen is chopped up white with a small black border on the left:

[image]

Jelos runs fine on the built-in display (and the HDMI).

I've ordered my RGB30 in the end of January. I could provide photos of the PCB and other components if needed.

### How can the issue be reproduced? (If applicable)

### What device are you using? (Only the Gameforce Chi, RK2020, OGA 1.0, OGA 1.1, RG351MP, RG351V, RGB10, R35s, RG353V/VS, RG353M, RG503, RGB30, and RK2023 units are officially supported)

RGB30

### What version of ArkOS are you on?

v2.0 10162023

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2024-02-25:
The only way that is possibly is because you loaded the wrong image for you device.  Make sure you’re loading the one for the RGB30.  The name RGB30 should be in the name of the downloaded image as well.

mx-bernhard, 2024-02-25:
And you are totally right. I am so sorry for bothering you. I accidentally had downloaded the image for the RK2023.

closed 2024-02-25 by mx-bernhard