christianhaitian/arkos #970: This one seems to work for the r35s per limited testing from a person on the AmberElec discord.

InZaineMind, 2024-02-25:
This one seems to work for the r35s per limited testing from a person on the AmberElec discord.
[rk3326-rg351mp-linux.zip](https://github.com/christianhaitian/arkos/files/12553858/rk3326-rg351mp-linux.zip)

_Originally posted by @christianhaitian in https://github.com/christianhaitian/arkos/issues/782#issuecomment-1710844140_

christianhaitian, 2024-02-25:
Hi,

Use the ArkOS image available from AeolusUX.  They are customized for the R3xs devices.
https://github.com/AeolusUX/ArkOS-R3XS

closed 2024-02-25 by christianhaitian

InZaineMind, 2024-02-25:
Hi, still the same black screen, should I change the DTB file same with the one being used with rf351mp IMage?

christianhaitian, 2024-02-25:
With the AeolusUX image, You should not need to change anything, just flash the image to the sd card then boot it

InZaineMind, 2024-02-26:
Still same. Black screen. :(

The one that makes a difference is the RG351MP but even when I change the DTB file it still shows white screen only.