christianhaitian/arkos #973: Feature: new theme

dani7959, 2024-02-27:
I made a theme for myself, but I ended up putting in quite some hours, so I decided to share it
https://github.com/dani7959/es-theme-replica/

It's pixel perfect, at least on 640x480 devices. 1:1 and 16:9 should also be, but I cannot check.

Have a look when you get bored, see if you consider it fit for ArkOS ;-)

christianhaitian, 2024-02-27:
Very cool.  zxspectrum is missing.  @JohnIrvine1433 how best can we have this theme added to ThemeMaster?

dani7959, 2024-02-27:
Looks like a partial initial upload, I will upload the rest in the morning.
Everything in the es_systems.cfg is themed except for retropie (options), vhs (movies) and ports.

lauren7ino, 2024-02-28:
👍🏼

JohnIrvine1433, 2024-02-28:
I can add @dani7959 repository to ThemeMaster but as there is only one theme best way forward would be to request Jetup to add it to https://github.com/Jetup13/Emulationstation-OGA-Theme-Gallery so it will be included in ThemeMaster's gallery mode.

christianhaitian, 2024-02-28:
Ok.  I've asked Jetup to do this.  Thanks!

christianhaitian, 2024-02-28:
Thanks to Jetup this theme has now been added to the Emulationstation Theme Gallery.  Thanks @dani7959 and @JohnIrvine1433 for the theme and guidance for inclusion into thememaster respectively.

lauren7ino, 2024-03-02:
Hi
I can't find this theme on the Theme Gallery of Theme Master.
What is the name I must search for?
Thanks

JohnIrvine1433, 2024-03-02:
It will be under replica.

lauren7ino, 2024-03-03:
"Will" means future? 
I searched and I can't find it anywhere in the app (found it on in Github though)

JohnIrvine1433, 2024-03-05:
No it will be automatically added as soon as you update the gallery (unless you've filtered the gallery by tested on 480x320 as the theme is marked as untested)