christianhaitian/arkos #980: default user/pass invalid credentials

fexofenadine, 2024-03-02:
### Describe the Issue (If applicable)
Default username password (ark/ark) does not authenticate through filebrowser web frontend or ssh

### How can the issue be reproduced? (If applicable)
Enter username: ark and password: ark at ssh password prompt or http://[ip address]:80

### What device are you using?
RG353V

### What version of ArkOS are you on?
ArkOS 2.0 (022320244)

### Anything else you'd like to include that may help to review this issue or feature request?
Using the password reset (Change Password) function in the Configuration menu results in "Password change failed". For all intents, it appears that either the username ark, or the password ark is not the documented default.  I have never changed the password to my knowledge, I've never successfully authenticated via ssh or web. I found reference to the combo root/351elec somewhere but that doesn't authenticate either, naturally.

closed 2024-03-02 by fexofenadine

christianhaitian, 2024-03-03:
user ark: and passwd: ark are always the default authentication.  Unless the image you're using was one not obtained from the wiki linked downloads.  If the image is one that you obtained through some other unsupported means, I can't help you.

fexofenadine, 2024-03-03:
thank you for the kind response, the fault was completely mine, I had changed the password and forgotten that I even did it.

apologies for wasting your time mate