christianhaitian/arkos #981: Install Headers issue

DingirAn, 2024-03-03:
Describe the Issue

When trying to install the Headers (https://github.com/christianhaitian/arkos/blob/main/Headers/README.md), the device gives a "/home/ark/.config/.DEVICE no such file or directory" error

### How can the issue be reproduced? (If applicable)

RG351MP

### What version of ArkOS are you on?

2.0 (02232024)

### Anything else you'd like to include that may help to review this issue or feature request?

If a file called .DEVICE is added on that path, the thing just shows a blinking cursor black screen, but no advance at all after that point.

christianhaitian, 2024-03-03:
What is your device?

DingirAn, 2024-03-03:
RG351MP

christianhaitian, 2024-03-03:
The issue has now been resolved.  You might still see the `/home/ark/.config/.DEVICE no such file or directory error` but the process should still continue without issue.  Thanks for reporting this.

closed 2024-03-03 by christianhaitian